openwrt/packages #6863: shorewall: busybox find with "newer" required

wjohnson-metroland, 2018-08-25:
Minor issue but please note that if you are building the latest version of ShoreWall 5.2.0.5 on OpenWRT you must configure BusyBox "find" binary with the "newer" feature enabled.

Default is:

```
config BUSYBOX_DEFAULT_FEATURE_FIND_NEWER
	bool
	default n
```

Note the default set to "n".

Ensure you switch it to "y" otherwise your firewall rules will not load properly which may lead to bricking, etc. I had enough space to enable all find features but i believe the only extra feature required that's not enabled by default is "newer".

diizzyy, 2018-08-25:
Compare difference in size and submit a patch upstream (OpenWrt) including that information unless there's a huge difference in size.

wvdakker, 2018-09-03:
@wjohnson-metroland 

I did a new clean pull and menuconfig.
The default option I get CONFIG_BUSYBOX_DEFAULT_FEATURE_FIND_NEWER=y.

wjohnson-metroland, 2018-09-04:
Hi @wvdakker 

This is strange because I also just did a default build and it's not checked. Perhaps this has something to do with the base system configuration when the Target Profile is set to my router type? These were my steps to replicate:

1. Vagrant build of Ubuntu 16.04 (VM from scratch)
2. installed all prerequisite tools (build-essential, git, etc)
3. Install as follows:

`git clone -b chaos_calmer git://github.com/openwrt/chaos_calmer.git`
`./scripts/feeds update -a`
`./scripts/feeds install -a`
`make menuconfig`

4. I selected the Target System to be Atheros AR7xxx
5. I selected Target Profile (TP-Link TL-WDR4300)
6. Go to Base System --> busybox --> customize busybox options --> finding utilities and the following required option is not selected:

`
Enable -newer: compare file modification times (NEW)
`

[image]

wvdakker, 2018-09-04:
@wjohnson-metroland 

I followed your steps. Not for CC but for the latest stable OpenWrt 18.06.1.
The CONFIG_BUSYBOX_DEFAULT_FEATURE_FIND_NEWER is enabled.

I think the version you use (Chaos Chalmer) has a older Busybox configuration setting. Because CC is not maintained anymore the issue cannot be fixed. 
Perhaps you switch to a newer OpenWrt version. The router you use is well supported. If it is not possible to switch then you have to enable the Busybox option manually.

dibdot, 2018-09-04:
CC is no longer supported - close this by now.

closed 2018-09-04 by dibdot