openwrt/packages #6883: socat crashes on execution - socat_1.7.3.2-2_powerpc_8540 Assertion failed opt_crdly.arg3

mommmomm, 2018-08-26:
Maintainer: @\lynxis
Environment: mpc85xx (socat_1.7.3.2-2_powerpc_8540 and socat_1.7.3.1-1_powerpc_8540) on LEDE 17.01.5, r3919-38e704be71 (TP-Link 4900)

Description:
Socat powerpc_8540 crashes immediatly on execution:
```
root@LEDE:/tmp# socat
Assertion failed: 3 << opt_crdly.arg3 == CRDLY (xioinitialize.c: xioinitialize: 41)
Aborted
```

Google says, a similar bug might have existed earlier, having something to do with missing variables during build that lead to a successful build, but a defunct binary throwing exactly the mentiones Assertion.

https://dev.archive.openwrt.org/changeset/26728

https://stackoverflow.com/questions/37120685/cannot-configure-termios-in-socat-xioinitialize-assertion

Also the socat FAQ mentiones:
```
Q: On Solaris, socat was built successfully, but when started, an assertion
fails: "xioinitialize.c:25: failed assertion `3 << opt_crdly.arg3 == CRDLY'

A: Probably, in a second attempt you set the correct LD_LIBARY_PATH for socat,
but it had not been set during the ./configure run, or you did not "make clean"
before running configure. Try it again:
make distclean; ./configure; make
```

diizzyy, 2018-08-27:
Can you please try 18.X or master instead?

mommmomm, 2018-08-28:
It will take a while until I can get the device offline for a complete dist upgrade (and I tried the most current package of socat from 18.x with no different result, see above). Unfortunately, the device is in active use with some work spent on configuration - testing a regular package was worth the risk (backups exists), but voluntarily killing it is another story. I wont be able to do this within the next 3 weeks, sorry.

But: If if really is a similar bug to those mentioned above, it should perfectly trigger on any powerpc_8540 device using one of the two packages mentioned above by simply invoking socat with or without any parameters.

What I could do: As an experiment I ran socat on
- a current (Openwrt 18.x, socat pulled via opkg yesterday, should be 1.7.3.2-2) ARM Raspberry PI (1st gen)
- a LEDE 17.01.4 (socat_1.7.3.1-1_mips_24kc) TP-Link Archer C7 (MIPS 74Kc) 
Both run without errors, so this either seems to be something like a reintroduction of the historic error above or a weird error that only hits LEDE 17.01.5, but not 17.01.4. The socat package versions were the sames as tried on the bugged platform with system versions being pretty close both older and newer. 

I really hope someone with a current powerpc_8540 install, ideally mpc85xx / TP-Link 4900, could spare a moment to do "opkg install socat; socat; opkg remove socat" and post the result.

thess, 2018-08-29:
I believe the problem is related to the TERMIOS definitions for powerpc, sparc and alpha kernels which are different that all others. The good news is I believe a fix can be done in the socat builds and backported to 17.01.5. The bad news is I have no way of testing the fix since I have no ppc systems available.

thess, 2018-08-30:
Check out PR #6917. I will push this to master, 18.06 and 17.01 shortly if you cannot directly test the fix from the PR. You will have to wait for buildbots to catch up.

mommmomm, 2018-08-30:
That was quicker than expected, thank you! I´ll test it immediatly as soon as I´m back from my business trip (~2 weeks from now).

thess, 2018-09-18:
Ping @mommmomm -- have you tested the fix?

mommmomm, 2018-09-21:
Sorry for the delay, hiding on the other end of the world somehow piled quite a few tasks on my desk.  

Today I was able to test the current (snapshot) package on the same install as the one mentioned above (LEDE 17.01.4 TP-Link Archer C7 (MIPS 74Kc)):
https://downloads.openwrt.org/snapshots/packages/powerpc_8540/packages/socat_1.7.3.2-3_powerpc_8540.ipk
(Date: Sun Sep 16 06:07:24 2018)
(This should contain your patch PR #6917  right?)

From my perspective **the issue is solved**, these are the tests I did:

Basic execution:
socat
2018/08/27 09:39:03 socat[3221] E exactly 2 addresses required (there are 0); use option "-h" for help
--> i.O.

socat -V
socat by Gerhard Rieger and contributors - see www.dest-unreach.org
socat version 1.7.3.2 on 1537031922
   running on Linux version #0 Fri Jul 13 19:25:14 2018, release 4.4.140, machine ppc
--> i.O.

I also tested packet-forwarding with some address translation on a more complex commandline with success (same behavior as on arm/pi).

Thank you  for fixing this (even more for fixing it so quickly)!

Issue can be closed / solved.

thess, 2018-09-21:
Good news - thanks for the reply

closed 2018-09-21 by thess