openwrt/packages #6893: Problem with php7-mod-mysqli package

alexander154, 2018-08-28:
Hi!

My OpenWRT device is:

```
Model: TP-Link Archer C20 v1
Architecture: MediaTek MT7620A ver:2 eco:6
Firmware Version: OpenWrt 18.06.0 r7188-b0b5c64c22 / LuCI openwrt-18.06 branch (git-18.210.69179-6df9a57)
Kernel Version 4.14.54
```

I have installed next packages:

> php7 php7-cli php7-mod-msyqli php7-mod-mysqlnd php7-mod-hash

When I run a test script, it show me next:

```
root@LEDE:~# php-cli 1.php
PHP Warning:  PHP Startup: Unable to load dynamic library 'mysqlnd.so' (tried: /usr/lib/php/mysqlnd.so (Error relocating /usr/lib/php/mysqlnd.so: RSA_size: symbol not found), /usr/lib/php/mysqlnd.so.so (Error loading shared library /usr/lib/php/mysqlnd.so.so: No such file or directory)) in Unknown on line 0
PHP Warning:  PHP Startup: Unable to load dynamic library 'mysqli.so' (tried: /usr/lib/php/mysqli.so (Error relocating /usr/lib/php/mysqli.so: mysqlnd_reverse_api_register_api: symbol not found), /usr/lib/php/mysqli.so.so (Error loading shared library /usr/lib/php/mysqli.so.so: No such file or directory)) in Unknown on line 0

Fatal error: Uncaught Error: Class 'mysqli' not found in /root/1.php:2
Stack trace:
#0 {main}
  thrown in /root/1.php on line 2

```

mhei, 2018-08-28:
Yeah, already noticed this problem, but did not yet manage to free a time slot to code a proper fix and test the result :-)

closed 2018-09-02 by mhei

tony-sunjian, 2018-10-09:
Rename 20_mysqli.ini to 30_mysqli.ini, or delete 20_mysqli.ini if both exist. If you also have 20_openssl.ini there, rename that to 10_openssl.ini.

commit 5afeb3f9 (2018-09-02): php7: adjust load priority for openssl (fixes #6893)

This orders loading of openssl extension before extensions
which require openssl functions.
commit 85c61655 (2019-03-26): php7: adjust load priority for openssl (fixes #6893)

This orders loading of openssl extension before extensions
which require openssl functions.

(cherry picked from commit 5afeb3f9c8512d5d786766bd394c4e2f6371f99f)