openwrt/packages #6905: nginx failure with ipv6 disabled

lucize, 2018-08-29:
@Ansuel 
compiled a x86 target with nginx-luci support, but disabled ipv6 globally, the service will fail with
`nginx: [emerg] socket() [::]:80 failed (97: Address family not supported by protocol)`
I see in default config  `listen [::]:80 default_server;` is there a way to prevent this if ipv6 is not selected ?
do we need ipv6 support by default ?

neheb, 2020-01-12:
Is this still an issue?

lucize, 2020-01-12:
still, disable ipv6 overall an nginx will not start

neheb, 2020-01-22:
Can you post an strace log?

peter-stadler, 2020-01-22:
I think this is a configuration issue (removing the `listen [::] …` solves it, right?) I am on the way changing the default configuration (see #11081) to make it easier extensible and want to address also this.

lucize, 2020-01-22:
Yes, just a config, I think is useless to listen on ipv6 by default, who want it should set it

PR #11081 (2020-02-03): nginx: use /etc/nginx/nginx.conf enabling conf.d/
Maintainer: Thomas Heil heil@terminal-consulting.de and Ansuel Smith ansuelsmth@gmail.com
Compile tested: MIPS 74K, Asus RT-N16, master snapshot
Run tested: MIPS 74K, Asus RT-N16, master snapshot, let it serve luci-ssl-nginx and luci-nginx

Description:
Instead of the default `nginx.conf` file this file is a small variant without examples that enables the `/etc/nginx/conf.d/` directory. It will pull in all configuration files from the conf.d directory. So, other packages can add their `server {…}` parts in that directory without modifying the m [... cut]
closed 2020-01-22 by lucize