openwrt/packages #6915: perl modules dependencies

lucize, 2018-08-30:
@pprindeville as I see you are doing active work on Perl, can you help me understand if I'm doing something wrong ?

I would like to add some perl modules in hope I can add spamassassin or amavisd-new but some modules have dependencies on others
an (non fatal build) example:

Net::DNS 

`Warning: prerequisite Digest::HMAC 1.03 not found.`

I have previously build the Digest::HMAC and added the host build options that I saw in other perl modules (perl-dbi, perl-inline..), if I manually copy the files in perl hostpkg folder then the warning disappears

```
HOST_BUILD_DIR:=$(BUILD_DIR_HOST)/perl/Digest-HMAC-$(PKG_VERSION)

HOST_BUILD_DEPENDS:=perl/host

include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/host-build.mk
include ../perl/perlmod.mk
[..]
define Host/Configure
        $(call perlmod/host/Configure,,)
        endef

define Host/Compile
        $(call perlmod/host/Compile,,)
endef

define Host/Install
        $(call perlmod/host/Install,$(1),)
endef
[..]
$(eval $(call HostBuild))
```
the HMAC*.pm is not copied into `staging_dir/hostpkg/usr/lib/perl5/5.28.0/Digest` where the Net::DNS seems to look for, neither in `staging_dir/target-[arch]_musl/usr/lib/perl5/5.28/Digest`where Net::DNS compile is **not** interested to look for

is it normal for it to search in the hostpkg dir and not in the target-[arch] dir ?
do you think that perlmod.mk needs some corrections so at least the *.pm files get copied in the hostpkg dir  ? it seems that now, only perlbase modules are copied in hostpkg dir.
how could I debug why Host/[Configure/Compile/Install] does nothing ?

pprindeville, 2018-09-09:
Have you looked at `feeds/packages/lang/perl-*/Makefile`?

lucize, 2018-09-10:
yes, they have something like

> HOST_BUILD_DIR:=$(BUILD_DIR_HOST)/perl/Digest-HMAC-$(PKG_VERSION)
>
> HOST_BUILD_DEPENDS:=perl/host
>
> include $(INCLUDE_DIR)/package.mk
> include $(INCLUDE_DIR)/host-build.mk
> include ../perl/perlmod.mk [...]

but I think none of them have dependencies on perl-* packages, but only perl core packages that are getting copied in the staging dir

pprindeville, 2018-09-10:
No, you're looking in the wrong place.  Have a look at [perl-uri](https://github.com/openwrt/packages/blob/master/lang/perl-uri/Makefile) for instance.

neheb, 2020-01-12:
What's the progress on this?

lucize, 2020-01-12:
I think is the same but not interested in this any more

closed 2020-01-12 by lucize