openwrt/packages #6925: unbound: using DNS-over-TLS, ssl handshakes fail after a few days

zedinosaur, 2018-09-02:
Maintainer: @EricLuehrsen 
Environment: Linksys WRT1900ACSv1, mvebu arch, OpenWrt version: 18.06.1

Description:
I've configured unbound to use DNS-over-TLS for forwarding. Usually after the router has been up for a few days it starts failing to resolve, apparently failing the ssl handshake. Restarting it immediately solves the problem but it returns again after a few days.

log messages
```
Sat Sep  1 17:47:29 2018 daemon.notice unbound: [24512:0] notice: ssl handshake failed 1.0.0.1 port 853
Sat Sep  1 17:47:29 2018 daemon.err unbound: [24512:0] error: ssl handshake failed crypto error:07064041:lib(7):func(100):reason(65)
Sat Sep  1 17:47:29 2018 daemon.notice unbound: [24512:0] notice: ssl handshake failed 9.9.9.10 port 853
Sat Sep  1 17:47:29 2018 daemon.err unbound: [24512:0] error: ssl handshake failed crypto error:07064041:lib(7):func(100):reason(65)
Sat Sep  1 17:47:29 2018 daemon.notice unbound: [24512:0] notice: ssl handshake failed 9.9.9.10 port 853
Sat Sep  1 17:47:29 2018 daemon.err unbound: [24512:0] error: ssl handshake failed crypto error:07064041:lib(7):func(100):reason(65)
Sat Sep  1 17:47:29 2018 daemon.notice unbound: [24512:0] notice: ssl handshake failed 1.1.1.1 port 853
Sat Sep  1 17:47:29 2018 daemon.err unbound: [24512:0] error: ssl handshake failed crypto error:07064041:lib(7):func(100):reason(65)
```

unbound_ext.conf:
```
forward-zone:
  name: "."
  forward-addr: 1.1.1.1@853#cloudflare-dns.com
  forward-addr: 1.0.0.1@853#cloudflare-dns.com
  forward-addr: 9.9.9.10@853#dns-nosec.quad9.net
  forward-ssl-upstream: yes
```

unbound_srv.conf
```
do-tcp: yes
prefetch: yes
qname-minimisation: yes
rrset-roundrobin: yes
use-caps-for-id: yes
tls-cert-bundle: "/etc/ssl/certs/ca-certificates.crt" # For OpenWrt
```

EricLuehrsen, 2018-09-03:
OpenSSL is having fatal memory allocation errors at `/buffer.c: int BUF_MEM_grow(BUF_MEM *str, size_t len)`. I am unsure how to repeat your situation to capture it. Not much appears on the web. Generally however others have reported similar issues, but OpenSSL responses appear to be "unhelpful."

EricLuehrsen, 2018-09-03:
OpenSSL is having fatal memory allocation errors at `/buffer.c: int BUF_MEM_grow(BUF_MEM *str, size_t len)`. I am unsure how to repeat your situation to capture it. Not much appears on the web. Generally however others have reported similar issues, but OpenSSL responses appear to be "unhelpful." Try using one entry for only Cloudflare or only Quad9 and see what happens. Maybe one of the certificates introduces trouble. Try `qname-minimisation: no` and `rrset-roundrobin: no` which have little benefit with a fixed forwarding server. Your final query is always the whole query so the server knows all in the end.

EricLuehrsen, 2018-09-06:
Issues are converging. I would like to continue discussion in #6957 (one spot). Keep or close this issue as you see fit.

EricLuehrsen, 2018-09-22:
@dibdot or @hnyman can we close this as duplicate to #6957 where most of the feedback is now appearing?

dibdot, 2018-09-22:
Duplicate - let's focus on #6957.

closed 2018-09-22 by dibdot