openwrt/packages #6941: strangswan build issue

yangbolu1991, 2018-09-03:
Below patch introduced a build issue if enabled CONFIG_PACKAGE_strongswan.
2b3784cd strongswan: dont overwrite ipsec.conf and ipsec.user during upgrade

The ipsec.conf won't be found when installed it.
$(INSTALL_CONF) $(PKG_INSTALL_DIR)/etc/ipsec.conf $(1)/etc/

If CONFIG_PACKAGE_strongswan-mod-stroke was enabled together, the issue would gone.

Please help to fix. Thanks a lot.

dibdot, 2018-09-03:
Which OpenWrt version, which package release? Also please always inform the maintainer if you raise a ticket (as mentioned in the ticket template): @stintel

dibdot, 2018-09-14:
No feedback by TO, close this by now.

closed 2018-09-14 by dibdot

oning67, 2019-11-11:
It doesn't seem like `strongswan` alone is much of a package. I paired it with `strongswan-default`, e.g.
```
CONFIG_PACKAGE_strongswan=y
CONFIG_PACKAGE_strongswan-default=y
```
and it built fine.