openwrt/packages #6957: unbound: DNS resolver process gradually consumes RAM until none left, stops resolving

skath, 2018-09-05:
# Issue template (remove lines from top till here)

Maintainer: @EricLuehrsen 
Environment: arm_cortex-a9_vfpv3, wrt3200acm, OpenWrt 18.06.0 r7188-b0b5c64c22

Description: the unbound DNS resolver gradually consumes RAM over the course of up to three weeks, until all 512M are consumed at which point DNS resolution stops working.

```
root@link:~# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='18.06.0'
DISTRIB_REVISION='r7188-b0b5c64c22'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3'
DISTRIB_DESCRIPTION='OpenWrt 18.06.0 r7188-b0b5c64c22'
DISTRIB_TAINTS=''
root@link:~# 
```
```
root@link:~# opkg info unbound
Package: unbound
Version: 1.7.3-2
Depends: libc, libopenssl, libunbound
Status: install ok installed
Architecture: arm_cortex-a9_vfpv3
Conffiles:
 /etc/config/unbound efedbbe21c74a37135e98d9344aba20a23f69a7dc3b581659f1ceabdb744c02a
 /etc/unbound/unbound.conf 4ec559b4a9331fa1af4aea21d4dfffe1111cb107aabc256048c363b29a560da2
 /etc/unbound/unbound_ext.conf e8b6b7e5d1a92385ca39ac47e9c763a44411ca17e294a5caee3c09117c486755
 /etc/unbound/unbound_srv.conf 62b6e7d5d67d146f98c1a4687742a7929e53217f98e9068bb547ca4fbbfd984b
Installed-Time: 1533529446

```
```
root@link:~# uptime
 13:49:53 up 14 days, 21:25,  load average: 0.00, 0.00, 0.00
root@link:~# grep ^Vm /proc/$(pidof unbound)/status
VmPeak:   315772 kB
VmSize:   315772 kB
VmLck:         0 kB
VmPin:         0 kB
VmHWM:    268760 kB
VmRSS:    268760 kB
VmData:   312472 kB
VmStk:       132 kB
VmExe:       172 kB
VmLib:      2904 kB
VmPTE:       316 kB
VmPMD:         0 kB
VmSwap:        0 kB

```
```
root@link:~# grep -v -e '^#' -e '^ +#' -e '^\t#' -e '^$' /etc/unbound/*.conf
/etc/unbound/unbound.conf:server:
/etc/unbound/unbound.conf:      verbosity: 1
/etc/unbound/unbound.conf:      username: "unbound"
/etc/unbound/unbound.conf:      directory: "/var/lib/unbound"
/etc/unbound/unbound.conf:      chroot: "/var/lib/unbound"
/etc/unbound/unbound.conf:      pidfile: "/var/run/unbound.pid"
/etc/unbound/unbound.conf:      num-threads: 1
/etc/unbound/unbound.conf:      msg-cache-slabs: 1
/etc/unbound/unbound.conf:      rrset-cache-slabs: 1
/etc/unbound/unbound.conf:      infra-cache-slabs: 1
/etc/unbound/unbound.conf:      key-cache-slabs: 1
/etc/unbound/unbound.conf:      interface: 0.0.0.0
/etc/unbound/unbound.conf:      interface: ::0
/etc/unbound/unbound.conf:      access-control: 0.0.0.0/0 allow
/etc/unbound/unbound.conf:      access-control: ::0/0 allow
/etc/unbound/unbound.conf:      outgoing-num-tcp: 1
/etc/unbound/unbound.conf:      incoming-num-tcp: 1
/etc/unbound/unbound.conf:      outgoing-port-permit: "10240-65335"
/etc/unbound/unbound.conf:      outgoing-range: 60
/etc/unbound/unbound.conf:      num-queries-per-thread: 30
/etc/unbound/unbound.conf:      msg-buffer-size: 8192
/etc/unbound/unbound.conf:      infra-cache-numhosts: 200
/etc/unbound/unbound.conf:      msg-cache-size: 100k
/etc/unbound/unbound.conf:      rrset-cache-size: 100k
/etc/unbound/unbound.conf:      key-cache-size: 100k
/etc/unbound/unbound.conf:      neg-cache-size: 10k
/etc/unbound/unbound.conf:      target-fetch-policy: "2 1 0 0 0 0"
/etc/unbound/unbound.conf:      harden-large-queries: yes
/etc/unbound/unbound.conf:      harden-short-bufsize: yes
/etc/unbound/unbound.conf:
/etc/unbound/unbound.conf:python:
/etc/unbound/unbound.conf:remote-control:
/etc/unbound/unbound_ext.conf:forward-zone:
/etc/unbound/unbound_ext.conf:  name: "."
/etc/unbound/unbound_ext.conf:  forward-addr: 1.0.0.1@853
/etc/unbound/unbound_ext.conf:  forward-ssl-upstream: yes
/etc/unbound/unbound_srv.conf:use-syslog: no
/etc/unbound/unbound_srv.conf:logfile: /mnt/sda1/unbound_log
/etc/unbound/unbound_srv.conf:log-queries: yes
/etc/unbound/unbound_srv.conf:log-replies: yes
```
```
root@link:~# unbound-control stats_noreset
thread0.num.queries=158407
thread0.num.queries_ip_ratelimited=0
thread0.num.cachehits=15381
thread0.num.cachemiss=143026
thread0.num.prefetch=0
thread0.num.zero_ttl=0
thread0.num.recursivereplies=143025
thread0.requestlist.avg=0.536707
thread0.requestlist.max=11
thread0.requestlist.overwritten=0
thread0.requestlist.exceeded=0
thread0.requestlist.current.all=1
thread0.requestlist.current.user=1
thread0.recursion.time.avg=0.024748
thread0.recursion.time.median=0.0290863
thread0.tcpusage=0
total.num.queries=158407
total.num.queries_ip_ratelimited=0
total.num.cachehits=15381
total.num.cachemiss=143026
total.num.prefetch=0
total.num.zero_ttl=0
total.num.recursivereplies=143025
total.requestlist.avg=0.536707
total.requestlist.max=11
total.requestlist.overwritten=0
total.requestlist.exceeded=0
total.requestlist.current.all=1
total.requestlist.current.user=1
total.recursion.time.avg=0.024748
total.recursion.time.median=0.0290863
total.tcpusage=0
time.now=1536183628.523561
time.up=1289732.424684
time.elapsed=1289732.424684
mem.cache.rrset=1710898
mem.cache.message=1272729
mem.mod.iterator=16504
mem.mod.validator=0
mem.mod.respip=0
histogram.000000.000000.to.000000.000001=3748
histogram.000000.000001.to.000000.000002=0
histogram.000000.000002.to.000000.000004=0
histogram.000000.000004.to.000000.000008=0
histogram.000000.000008.to.000000.000016=0
histogram.000000.000016.to.000000.000032=0
histogram.000000.000032.to.000000.000064=1
histogram.000000.000064.to.000000.000128=150
histogram.000000.000128.to.000000.000256=155
histogram.000000.000256.to.000000.000512=259
histogram.000000.000512.to.000000.001024=400
histogram.000000.001024.to.000000.002048=595
histogram.000000.002048.to.000000.004096=1501
histogram.000000.004096.to.000000.008192=9296
histogram.000000.008192.to.000000.016384=1960
histogram.000000.016384.to.000000.032768=68939
histogram.000000.032768.to.000000.065536=34762
histogram.000000.065536.to.000000.131072=14711
histogram.000000.131072.to.000000.262144=3451
histogram.000000.262144.to.000000.524288=1787
histogram.000000.524288.to.000001.000000=903
histogram.000001.000000.to.000002.000000=270
histogram.000002.000000.to.000004.000000=98
histogram.000004.000000.to.000008.000000=20
histogram.000008.000000.to.000016.000000=15
histogram.000016.000000.to.000032.000000=4
histogram.000032.000000.to.000064.000000=0
histogram.000064.000000.to.000128.000000=0
histogram.000128.000000.to.000256.000000=0
histogram.000256.000000.to.000512.000000=0
histogram.000512.000000.to.001024.000000=0
histogram.001024.000000.to.002048.000000=0
histogram.002048.000000.to.004096.000000=0
histogram.004096.000000.to.008192.000000=0
histogram.008192.000000.to.016384.000000=0
histogram.016384.000000.to.032768.000000=0
histogram.032768.000000.to.065536.000000=0
histogram.065536.000000.to.131072.000000=0
histogram.131072.000000.to.262144.000000=0
histogram.262144.000000.to.524288.000000=0
num.query.type.A=132061
num.query.type.PTR=4393
num.query.type.MX=1
num.query.type.TXT=1774
num.query.type.AAAA=20040
num.query.type.SRV=132
num.query.type.DS=3
num.query.type.DNSKEY=3
num.query.class.IN=158407
num.query.opcode.QUERY=158407
num.query.tcp=0
num.query.tcpout=157574
num.query.ipv6=0
num.query.flags.QR=0
num.query.flags.AA=0
num.query.flags.TC=0
num.query.flags.RD=158407
num.query.flags.RA=0
num.query.flags.Z=0
num.query.flags.AD=0
num.query.flags.CD=6
num.query.edns.present=1741
num.query.edns.DO=6
num.answer.rcode.NOERROR=154408
num.answer.rcode.FORMERR=0
num.answer.rcode.SERVFAIL=938
num.answer.rcode.NXDOMAIN=3060
num.answer.rcode.NOTIMPL=0
num.answer.rcode.REFUSED=0
num.answer.rcode.nodata=17092
num.query.ratelimited=0
num.answer.secure=0
num.answer.bogus=0
num.rrset.bogus=0
num.query.aggressive.NOERROR=0
num.query.aggressive.NXDOMAIN=0
unwanted.queries=0
unwanted.replies=0
msg.cache.count=6741
rrset.cache.count=7397
infra.cache.count=1
key.cache.count=0
num.query.authzone.up=0
num.query.authzone.down=0
```

skath, 2018-09-05:
root@link:~# uci export unbound
package unbound

config unbound
        option add_extra_dns '0'
        option add_local_fqdn '1'
        option add_wan_fqdn '0'
        option dhcp_link 'none'
        option dns64 '0'
        option domain_type 'static'
        option edns_size '1280'
        option hide_binddata '1'
        option localservice '1'
        option manual_conf '0'
        option protocol 'default'
        option rebind_localhost '0'
        option rebind_protection '1'
        option recursion 'default'
        option resource 'default'
        option root_age '9'
        option ttl_min '120'
        option validator '0'
        option extended_luci '1'
        option listen_port '1053'
        option extended_stats '1'
        option domain '<redacted>'
        option enabled '1'
        option unbound_control '1'
        list trigger_interface 'lan'
        list trigger_interface 'wan'

root@link:~#

EricLuehrsen, 2018-09-06:
This appears to relate to #6925 because the error code string is sniffing around a high candidate area. The root cause seems to ooze from OpenSSL own buffer (resize) allocations. Unbound would extend the time to failure if it kept TLS connections open for a minimum time (fewer of them), but at the core, I think OpenSSL has a memory leak.

I have a TL-Archer-C7v2 (ath79) running DoT as a long term test. It is growing in an unusual way. Normal recursion mode of Unbound has 15% RAM used for any indefinite use, but TLS after 4-5 days is 29%. It hasn't fallen over yet, but it is a matter of time I think.

- reference: `ssl handshake failed crypto error:07064041:lib(7):func(100):reason(65)`
- module: `openssl/crypto/buffer/buffer.c: int BUF_MEM_grow(BUF_MEM *str, size_t len);`

Possible interim containment action would be to add a cron job to restart Unbound daily at 0300, or  trim some fat in the configuration and stretch it longer:
```
# almost the same as default but more rigidly defined
option resource 'medium'
# reduce prefetch and other extra connection generating activity
option recursion 'passive'
# forward server knows whole query anyway, so minimization is of little use
option query_minimize '0'
# might as well remove this noise factor also
option protocol 'ip4_only'
```
Side note: DNSSEC can be enabled with or without TLS and no change in memory consumption.
```
option validator '1'
```

hnyman, 2018-09-06:
No direct help, but just an observation. 
Finding out if that leak is inside unbound or openssl itself might be the first step.
Test with mixing the current and previous versions of openssl and unbound, try to identify the culprit.

Luckily we seem have very few local patches here in Openwrt, so it likely originates upstream.

anomeome, 2018-09-06:
As another datapoint; using openssl 1.1.1-pre9 on a mamba, and 1.0.2p on a rango, both devices eventually exhibit the same failure

cotequeiroz, 2018-09-06:
I hope I can offer some help.  I've added libressl as an openssl-option, just like I did for the different openssl versions.  This was a quick attempt at it, so don't expect much, but  I've compile it, along with unbound.  I haven't run-tested them yet, but if anyone wish to, it is at https://github.com/cotequeiroz/openwrt-1/tree/libressl.

EricLuehrsen, 2018-09-07:
> Luckily we seem have very few local patches for unbound here in Openwrt, so it likely originates upstream.

Yeah. I am generally not fond of code level patches being the "_permanent-temporary_" solution. Test it, pretty it up, and ship it upstream.

EricLuehrsen, 2018-09-11:
1.0.2o repeats the error.

cotequeiroz, 2018-09-11:
I'm starting a test with libressl 2.7.4, using TLS to forward requests to 1.1.1.1.  Is there anything I should do to trigger the bug?

EricLuehrsen, 2018-09-12:
More look ups accelerate the bug put the router on the "realtime graphs - connections" page which also forces a lot of reverse lookup. Browse normally to exercise rapid growth.

OpenSSL 1.1.0 repeats the error.

cotequeiroz, 2018-09-12:
I'm running three instances, one with openssl 1.1.1 final, another with libressl 2.7.4, and another, also with libressl, but without the TLS enabled.  I'm forwarding lookups from my corporate mailserver to all three simultaneously, so they all have the same load.  I'll leave them running for a while and post the results.  I had to abort my first run -- I "accidentally" upgraded openssl, which  killed one of the processes, and they were both gaining size.  I'll post the results later this week, or when I decide it's pretty clear that they're growing.

cotequeiroz, 2018-09-12:
Well, after some 8 hours of testing, I think we can conclude that, unless all SSL libraries tested have a memory leak, the culprit is unbound.  Here's my results from the 3 processes handling the same queries for 8 hours:
### Test started at 2018-09-12 03:23 GMT
```
/proc/28396/stat
28396 (unbound-libress) S 1 28395 28395 0 -1 4194624 881 0 0 0 14 4 0 0 20 0 1 0 91059687 11288576 1316 2147483647 4194304 4371120 2143577536 0 0 0 0 4096 16391 1 0 0 18 0 0 0 0 0 0 4440028 4451092 9265152 2143579619 2143579678 2143579678 2143580129 0
grep ^Vm /proc/28396/status
VmPeak:    11024 kB
VmSize:    11024 kB
VmLck:         0 kB
VmPin:         0 kB
VmHWM:      5264 kB
VmRSS:      5264 kB
VmData:     7076 kB
VmStk:       136 kB
VmExe:       176 kB
VmLib:      3632 kB
VmPTE:        24 kB
VmPMD:         0 kB
VmSwap:        0 kB
/proc/28393/stat
28393 (unbound-openssl) S 1 28392 28392 0 -1 4194624 868 0 0 0 14 8 0 0 20 0 1 0 91059479 14295040 1323 2147483647 4194304 4363632 2141325520 0 0 0 0 4096 16391 1 0 0 18 0 0 0 0 0 0 4431836 4443055 7643136 2141326744 2141326803 2141326803 2141327338 0
grep ^Vm /proc/28393/status
VmPeak:    13960 kB
VmSize:    13960 kB
VmLck:         0 kB
VmPin:         0 kB
VmHWM:      5292 kB
VmRSS:      5292 kB
VmData:     7032 kB
VmStk:       136 kB
VmExe:       168 kB
VmLib:      6620 kB
VmPTE:        28 kB
VmPMD:         0 kB
VmSwap:        0 kB
/proc/28390/stat
28390 (unbound) S 1 28389 28389 0 -1 4194624 877 0 0 0 16 1 0 0 20 0 1 0 91059265 11288576 1318 2147483647 4194304 4371120 2146959696 0 0 0 0 4096 16391 1 0 0 18 0 0 0 0 0 0 4440028 4451092 8785920 2146962912 2146962959 2146962959 2146963434 0
grep ^Vm /proc/28390/status
VmPeak:    11024 kB
VmSize:    11024 kB
VmLck:         0 kB
VmPin:         0 kB
VmHWM:      5272 kB
VmRSS:      5272 kB
VmData:     7076 kB
VmStk:       136 kB
VmExe:       176 kB
VmLib:      3632 kB
VmPTE:        24 kB
VmPMD:         0 kB
VmSwap:        0 kB
```
### As of 2018-09-12 11:45 GMT
```
/proc/28396/stat
28396 (unbound-libress) S 1 28395 28395 0 -1 4194624 3537 0 0 0 50757 3411 0 0 20 0 1 0 91059687 20148224 3630 2147483647 4194304 4371120 2143577536 0 0 0 0 4096 16391 1 0 0 18 0 0 0 0 0 0 4440028 4451092 9265152 2143579619 2143579678 2143579678 2143580129 0
grep ^Vm /proc/28396/status
VmPeak:    19676 kB
VmSize:    19676 kB
VmLck:         0 kB
VmPin:         0 kB
VmHWM:     14520 kB
VmRSS:     14520 kB
VmData:    15728 kB
VmStk:       136 kB
VmExe:       176 kB
VmLib:      3632 kB
VmPTE:        32 kB
VmPMD:         0 kB
VmSwap:        0 kB
/proc/28393/stat
28393 (unbound-openssl) S 1 28392 28392 0 -1 4194624 3694 0 0 0 32473 2955 0 0 20 0 1 0 91059479 26906624 4586 2147483647 4194304 4363632 2141325520 0 0 0 0 4096 16391 1 0 0 18 0 0 0 0 0 0 4431836 4443055 7643136 2141326744 2141326803 2141326803 2141327338 0
grep ^Vm /proc/28393/status
VmPeak:    26276 kB
VmSize:    26276 kB
VmLck:         0 kB
VmPin:         0 kB
VmHWM:     18344 kB
VmRSS:     18344 kB
VmData:    19348 kB
VmStk:       136 kB
VmExe:       168 kB
VmLib:      6620 kB
VmPTE:        40 kB
VmPMD:         0 kB
VmSwap:        0 kB
/proc/28390/stat
28390 (unbound) S 1 28389 28389 0 -1 4194624 967 0 0 0 1987 2082 0 0 20 0 1 0 91059265 11599872 1402 2147483647 4194304 4371120 2146959696 0 0 0 0 4096 16391 1 0 0 18 0 0 0 0 0 0 4440028 4451092 8785920 2146962912 2146962959 2146962959 2146963434 0
grep ^Vm /proc/28390/status
VmPeak:    11328 kB
VmSize:    11328 kB
VmLck:         0 kB
VmPin:         0 kB
VmHWM:      5608 kB
VmRSS:      5608 kB
VmData:     7380 kB
VmStk:       136 kB
VmExe:       176 kB
VmLib:      3632 kB
VmPTE:        24 kB
VmPMD:         0 kB
VmSwap:        0 kB
```

Ansuel, 2018-09-21:
can confirm this issue 
so nobody reported this to unbound?

s0m3f00l, 2018-09-21:
I threw one together add to that 

https://www.nlnetlabs.nl/bugs-script/show_bug.cgi?id=4157

s0m3f00l, 2018-09-21:
They asked if you @cotequeiroz   they are asking for you to run valgrind with the ssl test. Can you take a look at the unbound developers post? I simply put all your comments into my bug report.   Thanks

EricLuehrsen, 2018-09-21:
Please read way up there. OpenSSL was notified (openssl#7128). NLNetlabs was notified (bugzilla?4149). NLNetlabs has been helping to hunt the leak for a few weeks.

cotequeiroz, 2018-09-21:
I’m on vacation until next week. I couldn’t run it under openwrt, even with debugging builds of OpenSSL and unbound. I will run it on x86_64 when I come back.

EricLuehrsen, 2018-09-21:
Note. Valgrind crashes hard almost instantly on MIPS. I dont think it likes the confusion "-Os" causes, necessary for most small targets though, and must be clever compact instructuon sequences it doesnt have a model for ir something like that.

cotequeiroz, 2018-09-25:
I've looked into the valgrind crash.  I've compiled valgrind, openssl and unbound with `-Og` instead of `-Os`, no stripping, and it fails at the same point. It seems to be related to MIPS64 assembly instructions being present in MIPS32 object files.  It may be trying to execute code that's not executable, or some confusion even with `-Og`.  What I've seen is that valgrind fails an assertion trying to execute MIPS64 instructions (CPU detection code?).  I'm not really familiar with the way valgrind works, so I may be wrong.  When (if) I get the time, I will try to run it on x86_64.

Ansuel, 2018-09-25:
Mvebu here and same problem... I don't think is related to mips64
instructions

EricLuehrsen, 2018-09-27:
It appears the Unbound leak will be resolved soon. Tests on select patches show promising results so far. Need to let this simmer for a while and see if it solved the growth or only slowed it.