openwrt/packages #6958: jamvm: Package missing for aarch64_cortex-a53 (bcm2710: Raspberry Pi 3 B)

sindastra, 2018-09-06:
Maintainer: @uxgood @champtar @danak6jq 
Environment: aarch64_cortex-a53, Raspberry Pi 3 B, 18.06.1

Description:

Package missing for aarch64_cortex-a53 (bcm2710: Raspberry Pi 3 B)

yousong, 2018-09-27:
@uxgood , I just cherry-picked the commit, but it's not enough.   The code in master branch already failed for different reasons.

https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/jamvm/compile.txt
```
checking whether build environment is sane... yes
checking for aarch64-openwrt-linux-strip... aarch64-openwrt-linux-musl-strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... aarch64-openwrt-linux-gnu
configure: error: aarch64-openwrt-linux-gnu not supported
Makefile:67: recipe for target '/var/lib/buildbot/slaves/slave-lede-builds4/aarch64_cortex-a53/build/sdk/build_dir/target-aarch64_cortex-a53_musl/jamvm-2.0.0/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/var/lib/buildbot/slaves/slave-lede-builds4/aarch64_cortex-a53/build/sdk/build_dir/target-aarch64_cortex-a53_musl/jamvm-2.0.0/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
time: package/feeds/packages/jamvm/compile#8.62#0.27#14.19
```

yousong, 2018-09-27:
Close this as it's an upstream issue, i.e. jamvm currently does not support aarch64.

http://jamvm.sourceforge.net/
```
The following platforms/architectures are recognised by configure. Those marked with * must be configured to use libffi.

Linux: x86, x86_64, ARM, PowerPC, PowerPC64(*), MIPS, HPPA
FreeBSD: x86, x86_64, ARM, PowerPC, PowerPC64(*), SPARC(*)
OpenBSD: x86, x86_64, ARM, PowerPC, PowerPC64(*), SPARC(*)
Mac OS X/Darwin: x86, x86_64, ARM, PowerPC, PowerPC64
Solaris/OpenSolaris: x86, x86_64
KFreeBSD: x86
```

closed 2018-09-27 by yousong