openwrt/packages #6977: ddns-scripts_route53-v1: fails to start

mark0n, 2018-09-08:
### Maintainer
@chris5560, @maxberger 

### Environment
OpenWRT 18.06.1

### Description
DDNS script for AWS Route 53 updates the IP successfully but then aborts with the following error:
```
root@LEDE:~# /usr/lib/ddns/dynamic_dns_updater.sh -S foo -v 1 -- start
 161133       : ************ ************** ************** **************
 161134  note : PID '8205' started at 2018-09-08 16:11
 161134       : ddns version  : 2.7.8-1
 161134       : uci configuration:
ddns.foo.check_interval='5'
ddns.foo.domain='<zone ID>'
ddns.foo.enabled='1'
ddns.foo.interface='wan'
ddns.foo.ip_interface='eth1'
ddns.foo.ip_source='interface'
ddns.foo.lookup_host='host.example.com'
ddns.foo.password='<AWS secret key ID>'
ddns.foo.service_name='route53-v1'
ddns.foo.use_https='1'
ddns.foo.username='<AWS access key ID>'
ddns.foo=service
 161138       : verbose mode  : 1 - run normal, console mode
 161138       : check interval: 300 seconds
 161139       : force interval: 259200 seconds
 161139       : retry interval: 60 seconds
 161139       : retry counter : 0 times
 161139       : No old process
 161140       : last update: never
 161140       : Detect registered/public IP
 161140       : #> /usr/bin/host -t A host.example.com  >/var/run/ddns/foo.dat 2>/var/run/ddns/foo.err
 161141       : Registered IP '98.209.41.42' detected
 161142  info : Starting main loop at 2018-09-08 16:11
 161142       : Detect local IP on 'interface'
 161143       : #> ip -o addr show dev eth1 scope global >/var/run/ddns/foo.dat 2>/var/run/ddns/foo.err
 161144       : Local IP '98.209.41.44' detected on interface 'eth1'
 161144       : Update needed - L: '98.209.41.44' <> R: '98.209.41.42'
 161144       : parsing script '/usr/lib/ddns/update_route53_v1.sh'
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   732  100   275  100   457    568    944 --:--:-- --:--:-- --:--:--  1802
 161145       : <?xml version="1.0"?>
<ChangeResourceRecordSetsResponse xmlns="https://route53.amazonaws.com/doc/2013-04-01/"><ChangeInfo><Id>/change/CL7GBU2VN2BC2</Id><Status>PENDING</Status><SubmittedAt>2018-09-08T16:11:45.884Z</SubmittedAt></ChangeInfo></ChangeResourceRecordSetsResponse>
 161146  WARN : PID '8205' exit WITH ERROR '1' at 2018-09-08 16:11

```
`/etc/config/ddns` contains
```
config service 'foo'
        option enabled '1'
        option service_name 'route53-v1'
        option use_https '1'
        option lookup_host 'foo.example.com'
        option domain '<zone ID>'
        option username '<AWS access key ID>'
        option password '<AWS secret key ID>'
        option ip_source 'interface'
        option interface 'wan'
        option ip_interface 'eth1'
        option check_interval '5'
```
I also have some entries for dyn.com which are working fine.

### Package versions
```
root@LEDE:~# opkg list-installed | egrep "ddns|curl|ssl"
curl - 7.60.0-3
ddns-scripts - 2.7.8-1
ddns-scripts_route53-v1 - 2.7.8-1
libcurl - 7.60.0-3
libopenssl - 1.0.2p-1
libustream-openssl - 2018-05-22-5322f9db-1
luci-app-ddns - 2.4.9-3
openssl-util - 1.0.2p-1
```
Note: I edited `/usr/lib/ddns/dynamic_dns_functions.sh` to apply my [fix for the curl SSL detection](https://github.com/openwrt/packages/pull/6747).

dibdot, 2018-09-09:
ddns-scripts is currently unmaintained. Maybe you'll find a fix on your own and provide an appropriate PR!?

dibdot, 2018-09-13:
@a-bali @mark0n please test proposed patch - thank you.

a-bali, 2018-09-13:
@dibdot Nope this does not help. First, $__MSG is not empty but contains a multi-line string like below, second, with this patch, $__MSE would remain unset if $__MSG is empty, throwing an error on the next line.

Content of $__MSG where urlencode fails:
` 121209       : <?xml version="1.0"?>
<ChangeResourceRecordSetsResponse xmlns="https://route53.amazonaws.com/doc/2013-04-01/"><ChangeInfo><Id>/change/xxxx</Id><Status>PENDING</Status><SubmittedAt>2018-09-13T10:12:09.864Z</SubmittedAt></ChangeInfo></ChangeResourceRecordSetsResponse>`

dibdot, 2018-09-13:
thanks for testing & sorry for the mess, I don't use it at all.

dibdot, 2018-09-13:
@a-bali another approach. I've replaced the shell url encoder with a awk variant (which should produce the same output but much, much quicker and hopefully without the mentioned bug). It's a drop in replacement of urlencode function ... maybe you give it a try ... ;-)
```
urlencode() {
	# $1	Name of Variable to store encoded string to
	# $2	string to encode
	local __STR __CHAR __OUT
	local __ENC=""
	local __POS=1

	[ $# -ne 2 ] && write_log 12 "Error calling 'urlencode()' - wrong number of parameters"

	__STR="$2"		# read string to encode

	__ENC="$(awk -v url="$__STR" 'BEGIN{ORS="";for(i=32;i<=127;i++)lookup[sprintf("%c",i)]=i
		for(i=1;i<=length(url);++i){enc=substr(url,i,1);if(enc!~"[-_.~a-zA-Z0-9]")enc=sprintf("%%%x", lookup[enc]);print enc}}')"

	eval "$1=\"$__ENC\""	# transfer back to variable
	return 0
}
```

a-bali, 2018-09-13:
@dibdot This appears to work fine, thanks. No idea why the previous version was not working though in the script, it worked fine via command line.

dibdot, 2018-09-13:
@mark0n please test the proposed patch #7015 - thanks

mark0n, 2018-09-14:
Thanks for looking into this. A first quick test looks promising. However, the script now aborts when started automatically after booting the system. It works fine when I press the "start" button on the web GUI. Need to investigate more.

dibdot, 2018-09-14:
Thanks for your feedback. I think it's unrelated to my change, but maybe you can replace the direct "awk" call with "/usr/bin/awk" and see if this oddity disappears.

dibdot, 2018-09-17:
@mark0n any news on this, did you find the culprit?

dibdot, 2018-09-18:
@mark0n @a-bali I've fixed the new urlencode approach (have forgotten to define a min. length for the hex replacement) and protect the answer string in the aws updater with double quotes - please test. Thanks!

a-bali, 2018-09-18:
@dibdot Does not fix the issue reported by @mark0n above, the script still fails on [this(https://github.com/dibdot/packages/blob/a99c7cbed64f9d61f33889f64cbb127ccf6f5da8/net/ddns-scripts/files/dynamic_dns_functions.sh#L289) line. Changing it to `eval $__CMD` eliminates the "not found" error but the script still terminates.

dibdot, 2018-09-18:
@a-bali that's odd - did you update the awk encode function with the latest changes in #7015?

a-bali, 2018-09-18:
> @a-bali that's odd - did you update the awk encode function with the latest changes in #7015?

I did, seemingly this is an unrelated issue

dibdot, 2018-09-18:
and you still receive the "not found" error?

a-bali, 2018-09-18:
Yes

> and you still receive the "not found" error?

Yes

dibdot, 2018-09-18:
OK, I'll quit here. Maybe others find a suitable solution, without proper testing this makes no sense. Sorry.

dibdot, 2018-09-19:
@mark0n @a-bali please contact me directly by mail (dev@brenken.org) for further testing & fixing the remaining issues - thanks.

maxberger, 2018-09-26:
Hello,

after updating to 18.06 I now ran into the same issue. (PID exit with error). Interestingly what caused the script for me to fail was the write_log line.  After removing:
  write_log 7 "${ANSWER}"
the script itself returned sucessfully. It is still not starting, so there is something else wrong.

mark0n, 2018-09-27:
@maxberger in the context of this ticket we found a bunch of issues which are fixed by #7015 and https://github.com/dibdot/packages/commits/ddns-fix, respectively (thanks a lot to @dibdot and @a-bali!). We are getting closer (the script keeps running now) but I think there is at least one more issue to fix (seems like the IP doesn't get updated anymore when I manually change the IP in the Route53 web front-end). I hope I find some time to do some more troubleshooting soon.

mark0n, 2018-09-28:
@dibdot, you commented out the [`set -euo pipefail` line](https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/update_route53_v1.sh#L10) @maxberger copied from [upstream](https://github.com/yuvadm/route53-ddns/blob/master/route53.sh#L21). Looking at the code I realized that we should definitely remove this line since `dynamic_dns_updater` [calls `send_update`](https://github.com/dibdot/packages/blob/master/net/ddns-scripts/files/dynamic_dns_updater.sh#L367) which in turn [sources `$update_script`](https://github.com/dibdot/packages/blob/master/net/ddns-scripts/files/dynamic_dns_functions.sh#L901) (in our case `update_route53_v1.sh`). This means this `set` command is not being run inside a sub shell but affects code execution in `dynamic_dns_updater.sh` and `dynamic_dns_functions.sh`!

In a similar way [setting the `IFS` variable](https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/update_route53_v1.sh#L11) is causing problems as well. It's unclear to me why this variable was being set in the first place. It seems like things are working fine with this line removed.

Please see #7114 for a version that works for me. Thank you all for your help!

dibdot, 2018-09-28:
@mark0n @a-bali final patchset now in #7115 ... last chance for further changes ... ;-))

maxberger, 2018-09-28:
Awesome! I'll try to verify this during the day today (eu time). So if you can hold off submitting for 24h you'll get an extra test data point :)

dibdot, 2018-09-28:
@maxberger take your time, appreciate your feedback!

maxberger, 2018-09-29:
@dibdot I can confirm: I have both scritps in the current version of #7115 running on my openwrt 18.06.0 box, and they work just fine. Updates work (both ipv4 and ipv6) , and the process stays running. Thanks for the fixes!