openwrt/packages #6982: mwan3: periodic packet loss since commit 46ed09408e554f182a688e0a7c47cdf85fc6c15d

thuehn, 2018-09-09:
Maintainer: @feckert
Environment: latest OpenWrt trunk from 9.9.2018, x86_64, PCengies APU 2c4

Description:
since commit 46ed09408e554f182a688e0a7c47cdf85fc6c15d wman3 does not work properly on my side, which has worked for several years before with 2x vdsl lines load balanced

troubleshootiing steps:
mwan3 forwards periodically some packets coming from the wan side to the wan side again at the mac layer, which is  wrong and leads to packet losses at the lan host

setup:
2x vdsl modems connected to eth1 and eth2 on the APU board, eth0 is lan side

Test case:
ping 8.8.8.8 from lan host

observation:
mwan3 gateway does forward about every 20th icmp packet reply from 8.8.8.8 not to the lan direction but to the MAC of the wan vdsl gateway, which leeds on the lan machine to packet loss, as the reply never arrives.

troubleshooting steps:
-reverting the latest mwan3 commit one-by-one, recompile mwa3 .. install downgraded mwan3 packet on the gateway node.
-by reverting the latest 3 commits up to commit 46ed09408e554f182a688e0a7c47cdf85fc6c15d, mwan3 works properly and the wrong forward decisions are gone .. no packet loss anymore on the lan machine

TCPDUMP output on eth1 (vdsl uplink 1) - icmp with seq #19 is lost from lan machine bbecause is got forwarded to the MAC of the vdsl wan gateway instead of the lan side:
```
2 > 5.102.161.1: ICMP echo request, id 42728, seq 17, length 64
21:07:09.219270 30:e4:db:26:67:3f > 00:40:63:f1:1d:15, ethertype IPv4 (0x0800), length 98: 5.102.161.1
 > 5.102.161.62: ICMP echo reply, id 42728, seq 17, length 64
21:07:10.164665 00:40:63:f1:1d:15 > 00:07:b4:00:be:01, ethertype IPv4 (0x0800), length 98: 5.102.161.6
2 > 5.102.161.1: ICMP echo request, id 42728, seq 18, length 64
21:07:10.194806 30:e4:db:26:67:3f > 00:40:63:f1:1d:15, ethertype IPv4 (0x0800), length 98: 5.102.161.1
 > 5.102.161.62: ICMP echo reply, id 42728, seq 18, length 64
21:07:11.169922 00:40:63:f1:1d:15 > 00:07:b4:00:be:01, ethertype IPv4 (0x0800), length 98: 5.102.161.6
2 > 5.102.161.1: ICMP echo request, id 42728, seq 19, length 64
21:07:11.187833 30:e4:db:26:67:3f > 00:40:63:f1:1d:15, ethertype IPv4 (0x0800), length 98: 5.102.161.1
 > 5.102.161.62: ICMP echo reply, id 42728, seq 19, length 64
21:07:11.187957 00:40:63:f1:1d:15 > 00:07:b4:00:be:01, ethertype IPv4 (0x0800), length 98: 5.102.161.1
 > 10.10.4.115: ICMP echo reply, id 42728, seq 19, length 64
```

Current Workaround:
git revert the last 3 mwan3 commits until commit 46ed09408e554f182a688e0a7c47cdf85fc6c15d and everything works without packet loss.

Any recommendations to get the troubleshooting further ?

Greetings Thomas

thuehn, 2018-09-10:
Hi,

Just installed the latest mwan3_2.7.0-1_all.ipk on my gateway node.
Packet loss happens again. Here is the output of the requested routing info:

````
oot@gateway ~ # ip r
efault via 5.102.161.1 dev eth1 proto static src 5.102.161.62 metric 1
efault via 5.102.161.1 dev eth2 proto static src 5.102.161.17 metric 2
.102.161.0/24 dev eth1 proto static scope link metric 1
.102.161.0/24 dev eth2 proto static scope link metric 2
0.0.0.0/8 dev wlan1 proto static scope link metric 20
0.0.0.0/8 dev wlan0 proto static scope link metric 60
0.10.2.0/24 via 10.10.88.3 dev eth0.10 metric 9 onlink
0.10.3.0/24 via 10.10.88.3 dev eth0.10 metric 9 onlink
0.10.4.0/24 via 10.10.88.4 dev eth0.10 metric 9 onlink
...
10.14.88.1 via 10.10.88.5 dev eth0.10 metric 9 onlink
10.14.88.2 via 10.10.88.5 dev eth0.10 metric 9 onlink
10.14.88.16 via 10.10.88.5 dev eth0.10 metric 9 onlink
10.14.88.17 via 10.10.88.5 dev eth0.10 metric 9 onlink
10.15.10.1 via 10.10.88.50 dev eth0.10 metric 9 onlink
10.99.99.5 via 10.10.88.6 dev eth0.10 metric 9 onlink
172.17.0.0/16 dev br-public proto static scope link metric 50
192.168.0.0/16 dev br-evergate proto static scope link metric 40
````

````
root@gateway ~ # ip r s t 1
default via 5.102.161.1 dev eth1
5.102.161.0/24 dev eth1 proto static scope link metric 1
5.102.161.0/24 dev eth2 proto static scope link metric 2
10.0.0.0/8 dev wlan1 proto static scope link metric 20
10.0.0.0/8 dev wlan0 proto static scope link metric 60
10.10.2.0/24 via 10.10.88.3 dev eth0.10 metric 9 onlink
10.10.3.0/24 via 10.10.88.3 dev eth0.10 metric 9 onlink
10.10.4.0/24 via 10.10.88.4 dev eth0.10 metric 9 onlink
10.10.5.0/24 via 10.10.88.4 dev eth0.10 metric 9 onlink
10.10.6.0/24 via 10.10.88.2 dev eth0.10 metric 9 onlink
10.10.8.0/24 via 10.10.88.3 dev eth0.10 metric 9 onlink
...
10.14.88.16 via 10.10.88.5 dev eth0.10 metric 9 onlink
10.14.88.17 via 10.10.88.5 dev eth0.10 metric 9 onlink
10.15.10.1 via 10.10.88.50 dev eth0.10 metric 9 onlink
10.99.99.5 via 10.10.88.6 dev eth0.10 metric 9 onlink
172.17.0.0/16 dev br-public proto static scope link metric 50
192.168.0.0/16 dev br-evergate proto static scope link metric 40
````

````
root@gateway ~ # ip r s t 2
default via 5.102.161.1 dev eth2
5.102.161.0/24 dev eth1 proto static scope link metric 1
5.102.161.0/24 dev eth2 proto static scope link metric 2
10.0.0.0/8 dev wlan1 proto static scope link metric 20
10.0.0.0/8 dev wlan0 proto static scope link metric 60
10.10.2.0/24 via 10.10.88.3 dev eth0.10 metric 9 onlink
10.10.3.0/24 via 10.10.88.3 dev eth0.10 metric 9 onlink
10.10.4.0/24 via 10.10.88.4 dev eth0.10 metric 9 onlink
10.10.5.0/24 via 10.10.88.4 dev eth0.10 metric 9 onlink
10.10.6.0/24 via 10.10.88.2 dev eth0.10 metric 9 onlink
...
10.14.10.4 via 10.10.88.5 dev eth0.10 metric 9 onlink
10.14.10.6 via 10.10.88.5 dev eth0.10 metric 9 onlink
10.14.10.7 via 10.10.88.5 dev eth0.10 metric 9 onlink
10.14.88.1 via 10.10.88.5 dev eth0.10 metric 9 onlink
10.14.88.2 via 10.10.88.5 dev eth0.10 metric 9 onlink
10.14.88.16 via 10.10.88.5 dev eth0.10 metric 9 onlink
10.14.88.17 via 10.10.88.5 dev eth0.10 metric 9 onlink
10.15.10.1 via 10.10.88.50 dev eth0.10 metric 9 onlink
10.99.99.5 via 10.10.88.6 dev eth0.10 metric 9 onlink
172.17.0.0/16 dev br-public proto static scope link metric 50
192.168.0.0/16 dev br-evergate proto static scope link metric 40
````

````
root@gateway ~ # ip rule
0:      from all lookup local
1001:   from all iif eth1 lookup 1
1002:   from all iif eth2 lookup 2
2001:   from all fwmark 0x100/0x3f00 lookup 1
2002:   from all fwmark 0x200/0x3f00 lookup 2
2061:   from all fwmark 0x3d00/0x3f00 blackhole
2062:   from all fwmark 0x3e00/0x3f00 unreachable
32766:  from all lookup main
32767:  from all lookup default
````

````
root@gateway ~ # ps | grep mwan
 8976 root      3492 S    /bin/sh /usr/sbin/mwan3rtmon
18780 root      3384 S    /bin/sh /usr/sbin/mwan3track encoline eth1 online 5.102.161.62 8.8.4.4 208.67.222.222
23757 root      3384 S    /bin/sh /usr/sbin/mwan3track encoline2 eth2 online 5.102.161.17 8.8.8.8 208.67.220.220
25646 root      3240 S    grep mwan
````

The obvious difference to the working mwan output from ip r s t 1 & 2 is that the mwan3_2.7.0-1_all.ipk tables have much more entries.

Let me know iof you need anything else for troubleshooting.
I will shwitch back to mwan3_2.6.18-1_all.ipk for the time been.

GReetings Thomas

ptpt52, 2018-09-10:
The output is as good as expected, and please post the output of
 iptables-save 

I think there is no problem.

thuehn, 2018-09-10:
So I have just stored the iptables rules for mwan_2.7 (https://pastebin.com/RBQpQ9QJ) and mwan_2.6.18 (https://pastebin.com/LPKjeiFH)

The diff shows that certain mwan3_ifaces_out are specified in version 2.6.18, but missing in 2.7 .. maybe a hint for the packet loss problem.

````
diff -a --suppress-common-lines -y iptables_mwan3_2.7.txt iptables_mwan3_2.6.18.txt 
> :mwan3_iface_out_encoline - [0:0]
> :mwan3_iface_out_encoline2 - [0:0]
> :mwan3_ifaces_out - [0:0]
> -A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_ifaces_out
> -A mwan3_iface_out_encoline -o eth1 -m mark --mark 0x0/0x3f00
> -A mwan3_iface_out_encoline2 -o eth2 -m mark --mark 0x0/0x3f0
> -A mwan3_ifaces_out -m mark --mark 0x0/0x3f00 -j mwan3_iface_
> -A mwan3_ifaces_out -m mark --mark 0x0/0x3f00 -j mwan3_iface_
````

Greetings Thomas

ptpt52, 2018-09-10:
@thuehn 
hi, can you try this PR https://github.com/openwrt/packages/pull/6990
it maybe something related

ptpt52, 2018-09-10:
When the main table changes and is detected by the mwan3rtmon process, it deletes the contents of routing tables 1, 2 and recreates the 1, 2 tables. In this process, there is a gap routing table that is emptied, causing packet forwarding pathfinding errors.

thuehn, 2018-09-10:
@ptpt52 
in your pr/6990 you use 'diff' which is not part of the busybox default selection ... so this would need a dependency to get it working!

Could you explain a bit background of your intention with all of this major changes to the routing table to t1 and t2 ? What would you like to achieve ?

Greetings Thomas

thuehn, 2018-09-10:
As we have a full mesh with ~100 nodes behind our gateway nodes, and the routing table is periodically updated by OLSR as routiing deamon ... the main table will change every couple of seconds in our case as the mesh network is quite is dynamic routing wise. ... so it seems to me that your new design will not fit properly in such a 'main routing tables changes constantly' case but the former design did

ptpt52, 2018-09-10:
> Could you explain a bit background of your intention with all of this major changes to the routing table to t1 and t2 ? What would you like to achieve ?

1. The data sent by the local does not follow the link selection policy(balanced) to select the link.
for example:

> root@OpenWrt:~# ip route
default via 10.0.0.1 dev eth0 proto static src 10.0.0.2 metric 31
default via 20.0.0.1 dev eth1 proto static src 20.0.0.2 metric 32
10.0.0.0/24 dev eth0 proto static scope link metric 31
20.0.0.0/24 dev eth1 proto static scope link metric 32
192.168.1.0/24 dev br-lan proto static scope link src 192.168.1.1

we have link eth0 and eth1, the local out data always goes out through eth0, never goes to eth1 even though the link policy is a balanced. this is not as expected.

so I rework on this changes to archive this goal.
It proved to solve these problems: https://github.com/openwrt/packages/issues/6455
https://github.com/openwrt/packages/issues/6551
https://github.com/openwrt/packages/issues/5055

ptpt52, 2018-09-10:
> in your pr/6990 you use 'diff' which is not part of the busybox default selection ... so this would need a dependency to get it working!

ok, I will fix it with a dependency.

thuehn, 2018-09-10:
so are you saying that the loadbalancing for any local process on the router itself does not work by design right now ?

Could'nt you just put local routing for 127.0.0.1 to a separate table ?

ptpt52, 2018-09-10:
dependency updated in PR https://github.com/openwrt/packages/pull/6990

> so are you saying that the loadbalancing for any local process on the router itself does not work by design right now ?

the load balancing works with the latest code, but not without my changes.

> Could'nt you just put local routing for 127.0.0.1 to a separate table ?

that does not do the tricks

ptpt52, 2018-09-10:
The main routing table is updated frequently for your scenario. so please help test it with the PR  https://github.com/openwrt/packages/pull/6990

feckert, 2018-09-10:
> so are you saying that the loadbalancing for any local process on the router itself does not work by design right now ?

@thuehn This has never worked. 
Packages from the router "generated by a local process" never used mwan3 policy based routing. This packages always get routed first see https://upload.wikimedia.org/wikipedia/commons/3/37/Netfilter-packet-flow.svg (green network layer, routing decision) mwan3 hooks are called in the mangle table prerouting and output table. Only forwarded (routed) packages were policy based routed. So the workaround was to use local_source "lan" which has some drawbacks.

thuehn, 2018-09-15:
@ptpt52 
I have just installed the latest mwan3 version with the grep substituting the diff.
It is running since 10min un the gateway without any loss or other issues. So it looks good so far.
Thx for troubleshooting!

Greetings Thomas

teqlee, 2018-09-16:
Hi.
Lastest ver has error when i'm Diagnostics MWAN Status.
When Check IP Rules:
```
Failed to execute call dispatcher target for entry '/admin/status/mwan/diagnostics_display/wan/check_rules'.
The called action terminated with an exception:
/usr/lib/lua/luci/controller/mwan3.lua:176: attempt to call global 'translatef' (a nil value)
stack traceback:
	/usr/lib/lua/luci/controller/mwan3.lua:176: in function </usr/lib/lua/luci/controller/mwan3.lua:84>
	(tail call): ?
```
When Check routing table:
```
Failed to execute call dispatcher target for entry '/admin/status/mwan/diagnostics_display/wan/check_routes'.
The called action terminated with an exception:
/usr/lib/lua/luci/controller/mwan3.lua:195: attempt to call global 'translatef' (a nil value)
stack traceback:
	/usr/lib/lua/luci/controller/mwan3.lua:195: in function </usr/lib/lua/luci/controller/mwan3.lua:84>
	(tail call): ?
```

feckert, 2018-09-22:
@thuehn I think we could close this issue its solved?

feckert, 2018-09-22:
@teqlee This is a luci issue so please open a issue in the luci repo
thanks

feckert, 2018-10-02:
@teqlee I think my PR https://github.com/openwrt/luci/pull/2198 will fix this

teqlee, 2018-10-12:
> 
> 
> @teqlee I think my PR [openwrt/luci#2198](https://github.com/openwrt/luci/pull/2198) will fix this

Fixed, tks @feckert

feckert, 2018-11-21:
@hnyman no reaction from issue creater and the problem themes to be solved
I think we could close this issue

closed 2018-11-21 by dibdot