openwrt/packages #6993: Unbound 1.7.3 build failed

nasbdh9, 2018-09-10:
Unbound build failed
Architecture mipsel_24kc
This is true with GCC 7 or GCC 8
...And I didn't choose this package when configuring

EricLuehrsen, 2018-09-11:
We will need something more descriptive to help you. Grep your `.config` file to see if any Unbound flags are set to yes. Also `~# make -j1 V=s` might yield some information.

nasbdh9, 2018-09-11:
I canceled the openvswitch related flags and everything works fine.
The openvswitch makefile updated a few days ago, maybe the problem is here.

nasbdh9, 2018-09-12:
This error occurs after choosing to build openvswitch
>checking for ECDSA_sign... no
configure: error: OpenSSL does not support ECDSA: please upgrade or rerun with --disable-ecdsa
Makefile:193: recipe for target '/home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/unbound-1.7.3/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/unbound-1.7.3/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/nasbdh9/openwrt/feeds/packages/net/unbound'
time: package/feeds/packages/unbound/compile#8.07#2.19#10.25
package/Makefile:107: recipe for target 'package/feeds/packages/unbound/compile' failed
make[2]: *** [package/feeds/packages/unbound/compile] Error 2
make[2]: Leaving directory '/home/nasbdh9/openwrt'
package/Makefile:103: recipe for target '/home/nasbdh9/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/nasbdh9/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/nasbdh9/openwrt'
/home/nasbdh9/openwrt/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

There is no error after setting the following values in .config

```
# CONFIG_PACKAGE_kmod-openvswitch is not set
# CONFIG_PACKAGE_kmod-openvswitch-geneve is not set
# CONFIG_PACKAGE_kmod-openvswitch-geneve-intree is not set
# CONFIG_PACKAGE_kmod-openvswitch-gre is not set
# CONFIG_PACKAGE_kmod-openvswitch-gre-intree is not set
# CONFIG_PACKAGE_kmod-openvswitch-intree is not set
# CONFIG_PACKAGE_kmod-openvswitch-lisp-intree is not set
# CONFIG_PACKAGE_kmod-openvswitch-stt-intree is not set
# CONFIG_PACKAGE_kmod-openvswitch-vxlan is not set
# CONFIG_PACKAGE_kmod-openvswitch-vxlan-intree is not set
```

dibdot, 2018-09-14:
this seems to be pretty unrelated to unbound - close this by now. Feel free to open a openvswitch related issue, if it's still a problem for you ... and please inform the openvswitch maintainer with @-notation, like mentioned in the issue template.

yousong, 2018-10-10:
The error message was from unbound configure script.  Open vSwitch 2.10 contains no mentions of `ECDSA_sign` while the code is present in unbound configure.ac

```
https://github.com/NLnetLabs/unbound/blob/master/configure.ac#L969

AC_ARG_ENABLE(ecdsa, AC_HELP_STRING([--disable-ecdsa], [Disable ECDSA support]))
use_ecdsa="no"
case "$enable_ecdsa" in
    no)
      ;;
    *)
      if test $USE_NSS = "no" -a $USE_NETTLE = "no"; then
	      AC_CHECK_FUNC(ECDSA_sign, [], [AC_MSG_ERROR([OpenSSL does not support ECDSA: please upgrade or rerun with --disable-ecdsa])])
```

If we look more closely, the posted error message also confirmed this

```
checking for ECDSA_sign... no
configure: error: OpenSSL does not support ECDSA: please upgrade or rerun with --disable-ecdsa
Makefile:193: recipe for target '/home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/unbound-1.7.3/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/unbound-1.7.3/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/nasbdh9/openwrt/feeds/packages/net/unbound'
time: package/feeds/packages/unbound/compile#8.07#2.19#10.25
package/Makefile:107: recipe for target 'package/feeds/packages/unbound/compile' failed
make[2]: *** [package/feeds/packages/unbound/compile] Error 2

```

yousong, 2018-10-10:
Ping unbound's current maintainer @EricLuehrsen ;)

yousong, 2018-10-10:
unbound is now at 1.8.0 on master.  Both unbound and openvswitch compile fine according to contents at downloads.openwrt.org [1,2]

If the issue still persists, you will need to provide OpenWrt version information and relevant bits in config.log which should reside in the PKG_BUILD_DIR.

 [1] https://downloads.openwrt.org/snapshots/faillogs/mipsel_24kc/packages/
 [2] http://downloads.lede-project.org/snapshots/packages/mipsel_24kc/packages/

EricLuehrsen, 2018-10-10:
Lets apply IS and IS-NOT. IS when `openvswitch` is installed. IS NOT any other `unbound` install I we know of. Conlude likely openvswitch Makefile sets a persistent switch or env variable that poisons other packages. Please close as not a Unbound issue.

yousong, 2018-10-11:
openvswitch selects libunound in commit 483e12a9ec82ae594f4686629cb85af67259482c, a change to support asynchronous dns resolving introduced in openvswitch-2.10

The then travis build checks and my local build/run tests all worked fine.  I just re-checked with the following config options and cannot reproduce the said issues.

```
CONFIG_TARGET_malta=y
CONFIG_TARGET_malta_be=y
CONFIG_TARGET_malta_be_Default=y
CONFIG_PACKAGE_openvswitch=y
```

@froodproton @nasbdh9 , please provide relevant bits of unbound's config.log to further debug this.

```
build_dir/target-mips_24kc_musl/unbound-1.8.1/config.log

configure:17764: checking for ECDSA_sign
configure:17764: mips-openwrt-linux-musl-gcc -o conftest -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/yunion/git-repo/openwrt/openwrt/build_dir/target-mips_24kc_musl/unbound-1.8.1:unbound-1.8.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -D_GNU_SOURCE  -I/home/yunion/git-repo/openwrt/openwrt/staging_dir/target-mips_24kc_musl/usr/include -I/home/yunion/git-repo/openwrt/openwrt/staging_dir/target-mips_24kc_musl/include -I/home/yunion/git-repo/openwrt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/usr/include -I/home/yunion/git-repo/openwrt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/include/fortify -I/home/yunion/git-repo/openwrt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/include  -I/home/yunion/git-repo/openwrt/openwrt/staging_dir/target-mips_24kc_musl/usr/include -L/home/yunion/git-repo/openwrt/openwrt/staging_dir/target-mips_24kc_musl/usr/lib -L/home/yunion/git-repo/openwrt/openwrt/staging_dir/target-mips_24kc_musl/lib -L/home/yunion/git-repo/openwrt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/usr/lib -L/home/yunion/git-repo/openwrt/openwrt/staging_dir/toolchain-mips_24kc_gcc-7.3.0_musl/lib -znow -zrelro  -L/home/yunion/git-repo/openwrt/openwrt/staging_dir/target-mips_24kc_musl/usr/lib conftest.c  -lcrypto >&5
configure:17764: $? = 0
configure:17764: result: yes
```

yousong, 2018-10-17:
cannot reproduce and no further response happened.  close this for now.

yousong, 2018-10-30:
@NeoRaider Thanks for looking into this.

It's openvswitch userspace packages having a runtime dependencies on libunbound.  kmod-openvswitch should not.

NeoRaider, 2018-10-30:
Ugh, the issue is more annoying than I thought at first: OPENSSL_WITH_EC (which --disable-ecdsa can workaround) is not the only required symbol: OPENSSL_WITH_DEPRECATED is necessary as well (and AFAICT we can't get around that one without patching libunbound or upgrading to OpenSSL 1.1).

The main issue here is that all dependencies on config symbols (like +@OPENSSL_WITH_DEPRECATED) are only effective when a package is enabled in .config. When libunbound and openssl are only built as build dependencies, there is no nice way to ensure the dependency is fulfilled.

I think I have found an acceptable workaround (essentially by creating a hidden negated symbol OPENSSL_NO_DEPRECATED), but a user could still break the build by
- explicitly enabling openssl and disabling OPENSSL_WITH_DEPRECATED
- having libunbound built as a build dependency (so a potential +@OPENSSL_WITH_DEPRECATED dependency would not be considered)

I'll make a few more tests and push my fixes in a moment, but I assume we'll have to revisit the whole build dependency subject at some point.

EricLuehrsen, 2019-03-31:
Unbound deprecated API were patched in PR #7295 and Unbound 1.9.1 upstream included in PR #8394. Can we close this? It is seems to be more of a build mechanics issue (setting .config).

neheb, 2019-03-31:
Closing. https://github.com/openwrt/packages/pull/8553 makes OpenSSL ECC support explicit.

closed 2019-03-31 by neheb

userABCD-ship-it, 2020-01-22:
Sorry for the newb question as I am new to openwrt. I am seeing this same issue with unbound 1.9.1. Reading this thread, it is not clear what the solution or build configuration is to get it to work. Any suggestions?