openwrt/packages #6997: nut-server:

mantinan, 2018-09-11:
Maintainer: @\cshoredaniel
Environment: mips_24kc tp-link TL-MR3420 OpenWrt 18.06.1, r7258-5eb055306f

Description:

Sometimes the riello driver won't load, that is a problem for wich I don't have much data, the kernel says the cable disconnects during boot, don't know why, I believe a similar thing can also happen on other models.

The problem however is that when it fails to load you cannot do a restart, I tried /etc/init.d/nut-server restart and it doesn't restart or end the process, it stays there waiting for something that doesn't happen :-(

It sounds weird to me the processes that are left after a normal start, I always see this processes:

 1402 root       976 S    /usr/sbin/upsd
 1417 root      1248 S    /lib/nut/riello_usb -a riello
 1421 root      1316 S    /bin/sh /etc/rc.common /etc/init.d/nut-server running
 1426 root      1200 S    flock 1000

On previous versions we didn't have a shell running or a flock around, so memory would be lower, and I could do a restart of nut in case something had gone wrong with nut.

Regards.

cshoredaniel, 2018-09-11:
This is fixed in master; once master is sufficiently tested hopefully my backport PR will be accepted.

mantinan, 2018-09-11:
Ok, I'll try to see if I can backport it and see if it works.

On the other hand there is the failure to start of the riello driver, which I've seen several times, and I have another router connected to a couple sais that use apcsmart driver and also seem to have failed at least once, are also these kind of failures fixed on master?

If they are not fixed I'll have to investigate it.

I'd like to suggest that the apcsmart driver is also built on official openwrt, having to do it by hand is quite a lot of work and the driver works well. Is there any reason for not building it? if not... where should I ask for this?

Thanks in advance.

cshoredaniel, 2018-09-11:
For the riello, it might be sort of fixed in master (that is when it 'shows up' again in the kernel it should hotplug.

Is the apcsmart a serial driver?  I don't have any serial UPSes to test and most routers aren't really prepared to deal with RS232 inputs so they're not built by default.  Perhaps do a build with/without and show the size difference?  If it's not too much in the common packages the I'm all for building by default.

cshoredaniel, 2018-09-12:
FYI the documentation for what's in and what's being worked on, is here: https://openwrt.org/docs/guide-user/services/ups/software.nut#nut_server

cshoredaniel, 2018-09-13:
I think I'll enable serial by default after double checking no nasty size surprises on e.g. nut-server (not expecting any).  I'll need some help making sure things work with the master initscripts and such though as I don't have any such hardware.

mantinan, 2018-09-13:
Ok, it is working great on our setup right now, no problem on testing whatever  you need for that build.

dibdot, 2018-09-14:
Problem seems to be fixed - close this by now.

closed 2018-09-14 by dibdot

cshoredaniel, 2018-09-14:
Are you using master (with working restart) or 18.06 (which doesn't have it yet)?  I'd like to be sure master hasn't broken serial.

mantinan, 2018-09-14:
I'm ussing 18.06 right now, but I have compiled master and I plan to test it next week, I'll report back then.

mantinan, 2018-09-25:
I solved the error by setting option statepath /var/run/nut on config upsd, but only upsd is run, no sign of the driver, in this case it was apcsmart.

I'm wondering thy did I have to set statepath on the first place :-?

And then, the problem with the driver, is specific to my driver (because it is serial) or will it happen on any other driver?

I'm still at the same place as my previous message.

cshoredaniel, 2018-09-26:
Apologies everybody I've gotten drawn into local politics and it's 
talking up a *lot* of time.  Election's on October 22 so if I haven't 
gotten to this by then (although hopefully I can make time before then) 
I should have time after that.

mantinan, 2018-09-27:
I was testing with a stable installation and master nut packages but as I was not sure if I had to pull any other package from master, I'm now testing with a newly installed machine running master.

When running master (OpenWrt SNAPSHOT, r8165-3fa7e62) I get exactly the same results as explained before, this time I'm using driver riello_usb.

I finally got this to run with a couple of tricks, the one I commented of setting statepath, and the other was fixing the configs. Like Ivanich explains on PR #7096 driver riello_usb doesn't recognice a series of variables, so I was getting...

Fatal error: 'offdelay' is not a valid variable name for this driver. 
Fatal error: 'ondelay' is not a valid variable name for this driver. 
Fatal error: 'pollfreq' is not a valid variable name for this driver.

After commenting all those on the init.d script still riello_usb was not running, this time it was because of the permissions on the /dev/bus/usb devices, allowing nut to read and write that finally fixed the problem and it now runs ok.

 I'll try to get this running into stable as well using only the nut packages from master and report back.

cshoredaniel, 2018-09-27:
Thank you for the feedback.  I've gotten drawn into local (municipal) 
politics and have have been quite busy with 'life' as well so although I 
will make an effort to get to this before October 22 election I may not 
have a lot of time before then.

mantinan, 2018-09-28:
Hi again!

I tried the new packages on stable with the same results as on master, needed the ption statepath /var/run/nut to avoid the mkdir problems, the fatal errors with the 3 variables, which are not valid for riello_usb or apcsmart either, and the permissions problems.

On the riello_usb the permissions are needed on /dev/bus/usb/... while the apcsmart (which is a serial driver) needs permissions on the serial port, in my case this was a usb to serial adapter, thus /dev/ttyUSBX.

After fixing those problems everything seems to work ok.

Regards.

cshoredaniel, 2018-10-10:
@mantinan FYI you can create /etc/hotplug.d/tty and add a hotplug script under it to set permissions on the serial port.

mantinan, 2018-10-19:
Hi!

I have now tested it with an APC UPS running serial apcsmart with this /etc/hotplug.d/tty/99-USB script for the permissions problem and everything runs ok:
#!/bin/sh

[ -n "$DEVNAME" -a -z "${DEVNAME%ttyUSB*}" -a "$ACTION" = "add" ] &&
  chown nut "/dev/$DEVNAME"

The only problem that remains is the user problem (upsd running as root as I explained already on the wrong bug, sorry about that, and that I'm pasting again here), that was with my config for riello usb:

I have downloaded master from cshoredaniel:pr-nut-more-fixes and compiled it.

I'm finding that even though I have set it to run as nut I get upsd running as root, this used to be user nut on -8 version, so -9 has a regression here.

On -8 a ps would tell:
1172 nut 2900 S /usr/sbin/upsd -D -u nut
On -9 I get:
1258 root 2896 S /usr/sbin/upsd -D

This is my nut_server config:
config driver 'riello'
option driver riello_usb
option port /dev/ttyUSB0
config listen_address
option address 172.22.54.49
option port 3493
config upsd 'upsd'
option maxage 15
option statepath /var/run/nut
option maxconn 1024
option runas nut

Regards.

cshoredaniel, 2018-10-19:
On 2018-10-19 7:56 a.m., mantinan wrote:
> Hi!
> 
> I have now tested it with an APC UPS running serial apcsmart with this 
> /etc/hotplug.d/tty/99-USB script for the permissions problem and 
> everything runs ok:
> #!/bin/sh
> 
> [ -n "$DEVNAME" -a -z "${DEVNAME%ttyUSB*}" -a "$ACTION" = "add" ] &&
> chown nut "/dev/$DEVNAME"

Unfortunately forcing the 'nut' user to ttyUSB is probably a non-starter 
as it NUT is not the only consumer of serial USB.  I am thinking though 
that it might be worthwhile to have a 'nut_server' option that such a 
script reads to determine whether to run or not (so it only comes into 
play for NUT users).

> The only problem that remains is the user problem (upsd running as root
> as I explained already on the wrong bug, sorry about that, and that I'm
> pasting again here), that was with my config for [...]

Ah, I'll have to document that better -- runas as is now per-driver not 
per NUT server (so if you do runas under 'config driver 'riello'' it 
should work.

mantinan, 2018-10-22:
Hi again!

I'd like you to note something, I was talking that the upsd was run as root, while on my config it says that upsd is run as nut. This is what I think is wrong.

On the other hand as I hadn't specified a user for the riello driver it is started as nut by default, which looks ok.

Regards.

cshoredaniel, 2018-11-25:
Thanks, that should be fix the recently merged PR

mantinan, 2018-11-27:
Hi!

I have upgraded to the latest snapshots available and upsd still runs as root.
This happens both if you don't devine a config upsd section and if you define it specifying a runas nut.

So this doesn't seem to be fixed yet.

If you want me to test anything, just tell me.

Regards.

cshoredaniel, 2018-12-05:
On 2018-11-27 7:15 a.m., mantinan wrote:
>
> Hi!
>
> I have upgraded to the latest snapshots available and upsd still runs 
> as root.
> This happens both if you don't devine a config upsd section and if you 
> define it specifying a runas nut.
>
> So this doesn't seem to be fixed yet.
>
> If you want me to test anything, just tell me.
>
Thank you for you report.  I forgot to push my changes for this. I'm 
rebasing and testing again and will push again.

Regards,

Daniel

cshoredaniel, 2018-12-06:
@mantinan Ok, so the problem wasn't what I thought, but I know have a solution and will be pushing (hopefully in the AM).

cshoredaniel, 2018-12-11:
@mantinan FYI I've posted PR: https://github.com/openwrt/packages/pull/7638 which ought to do must of what's now in master except the USB hotplug bits (as that's a new feature IMO).  If you can, please test and comment!

mantinan, 2019-01-11:
I have commented on PR: #7638 that at least for the riello this is not working like it should. Sorry about the bold fonts, those where root prompts O:-)

cshoredaniel, 2019-01-11:
@mantinan quick question: is master not working as well (it does for me and another user, but that's not necessarily enough of an indicator) or did I miss something in the backport process?

cshoredaniel, 2019-01-12:
@mantinan For reference I've essentially reverted the runas stuff for 18.06 because it was a new feature (for OpenWrt) that never properly worked in 18.06 series, and is non-essential.  It should, however, be working in master, including with serial USB (if the appropriate option is enabled in /etc/config/nut_server).

mantinan, 2019-01-14:
Hi!
I have the 18.06 stable devices working as nut without problems with a -8 version that I had compiled previously, so that's ok with me.

As for master.... I've got a device installed with current snapshot but I've had a lot of trouble...
1- I could not install a thing, as standard wget was trying to go out through IPv6 which I don't have, instead of the proxy I had specified on my environment. So I had to install wget and dependencies by hand and then I could use opkg to install nut.
2- I cannot test riello driver which is the ups I have here as the usb of the Riello doesn't work ok with the tp-link Archer C7.
3- Testing the apcsmart driver I got problems to make it set the proper permissions on ttyUSB0, so the riello driver wouldn't start. This is basically because I don't know how to configure enable_usb_serial, this is my current config, if you tell me what's wrong I'll try again:

config driver 'apc1rack'
        option driver apcsmart
        option port /dev/ttyUSB0
	option enable_usb_serial 1
config upsd 'upsd'
	option maxage 15
	option statepath /var/run/nut
	option maxconn 1024
	option runas nut
	option enable_usb_serial 1

adding a commented enable_usb_serial example on the default config file would be great.

Regards.

cshoredaniel, 2019-01-16:
FYI I'm working in this; discovered a silly mistake off the top (missing sourcing /lib/functions.sh which means config_load was not actually available, hence no effect from enable_usb_serial).

cshoredaniel, 2019-01-22:
@mantinan Finally got https://github.com/openwrt/packages/pull/8017 pushed!  I hope you can check it out.  I think it should be the final iteration needed.

mantinan, 2019-01-28:
I've upgraded today to dhe -15 packages as well as underlaying libusb and that stuff, I still don't get ttyUSB0 with the right permissions.

Like I said on the previous post, I don't know if that config file is ok or where to put the enable_usb_serial option, that's why as you can see on that post, I have it twice.

I've just upgraded the packages related to nut and usb without changing the config file or anything else. If you think I should upgrade any other thing or change something on that config file, please let me know, right now it just launches upsd as nut but the apc driver fails to start.

Regards.

PR #7592 (2018-12-06): nut: Optionally set serial USB to NUT user
When using serial-port based UPSes with NUT, it is handy to be able to
configure a USB serial port to have be set tot the NUT runas user, so
that NUT can access the serial port automagically.
Closes #6997

Maintainer: me / @cshoredaniel 
Compile tested: brcm2708, Raspberry Pi B+, current masters (openwrt, packages, luci)
Run tested: same arch/model as compile test.  Doesn't break anything but I don't have hardware to test that it actually works.
commit a5d06ce1 (2018-12-06): nut: Optionally set serial USB to NUT user

When using serial-port based UPSes with NUT, it is handy to be able to
configure a USB serial port to have be set tot the NUT runas user, so
that NUT can access the serial port automagically.
Closes #6997