openwrt/packages #6998: stubby: image-builder doesn't find stubby

bogi788, 2018-09-11:
Maintainer: iamperson347+public@gmail.com
Environment: mips_24kc, TP-Link Archer C7 v2, OpenWRT Development Snapshot

Description: The image builder currently fails to build my custom image (with kernel modules and packages for my LTE modem) and stubby for DNS over TLS and DNSSEC. The error message is

```Unknown package 'stubby'.```

dibdot, 2018-09-14:
stubby is currently broken for your arch, see: https://downloads.openwrt.org/snapshots/faillogs/mips_24kc/packages/stubby/compile.txt, therefore it's not included in image builder.

bogi788, 2018-09-14:
Apparently this is caused by the getdns dependency also failing to build: https://downloads.openwrt.org/snapshots/faillogs/mips_24kc/packages/getdns/compile.txt

bogi788, 2018-09-18:
The getdns problem is probably related to https://github.com/openwrt/packages/pull/6960

adagari, 2018-09-22:
I am also having errors on mvebu when trying to use the image builder.

yousong, 2018-09-25:
The getdns missing dependency on libbsd issue should now be worked around by pull request #6960

closed 2018-09-25 by yousong