openwrt/packages #6999: mc - missing virtual file system support

ThomasCr, 2018-09-11:
Hi,
today I wanted to use mc to browse a tar archiv to extract only some files. I know it is possible to list content with tar command and than extract it manually, but with many files it feels not really comfortable.

thats why I use mc for those tasks, for easer extraction... problem is, that mc is not build with any Virtual File System, and so I can't open 

Is there a way to add it in oncoming releases, maybe in a extra package (aka mc-full) ?

Thanks,
Thomas

dibdot, 2018-09-12:
Enabling VFS support adds round 40KB to resulting binary (tested on mips target, 390KB vs. 433KB). Maybe it doesn't matter much to enable this by default cause it's already a big package ... any thoughts?

DazzyWalkman, 2018-09-12:
VFS support adds libtirpc dependency, that's another few KBs. 
To make it more convenient for packages repo users, I will think about adding an extra package (aka mc-full) , as OP suggests.  
For those who build packages for themselves, like me, leaving config options in menuconfig is already good enough.

mhei, 2018-09-12:
Maintainer decided to enable VFS support by default, so I think we can close this issue.

closed 2018-09-12 by mhei