openwrt/packages #7000: Boost not building on mips32/brcm63xx

thess, 2018-09-11:
Maintainer: @ClaymorePT 
Environment: mips_mips32/brcm63xx 18.06

Odd problem - I think ASM "pause" should only be generated for x86 ?

```
"mips-openwrt-linux-gcc"   -Os -pipe -mno-branch-likely -mips32 -mtune=mips32 -fno-caller-saves
-fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -I/
opt/ledesrc/staging_dir/target-mips_mips32_musl/usr/lib/libiconv-stub/include -I/opt/ledesrc/staging
_dir/target-mips_mips32_musl/usr/lib/libintl-stub/include -iremap/opt/ledesrc/build_dir/target-mips_
mips32_musl/boost_1_68_0:boost_1_68_0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_
SOURCE=1 -Wl,-z,now -Wl,-z,relro   -DBOOST_NO_FENV_H -fPIC  -std=gnu++14 -pthread -O3 -finline-funct
ions -Wno-inline -Wall -Os -pipe -mno-branch-likely -mips32 -mtune=mips32 -fno-caller-saves -fno-plt
 -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -I/opt/lede
src/staging_dir/target-mips_mips32_musl/usr/lib/libiconv-stub/include -I/opt/ledesrc/staging_dir/tar
get-mips_mips32_musl/usr/lib/libintl-stub/include -iremap/opt/ledesrc/build_dir/target-mips_mips32_m
usl/boost_1_68_0:boost_1_68_0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1
 -Wl,-z,now -Wl,-z,relro   -DBOOST_NO_FENV_H -fPIC  -DBOOST_ALL_NO_LIB=1 -DBOOST_DISABLE_ASSERTS -DB
OOST_FIBERS_SOURCE -DBOOST_FILESYSTEM_STATIC_LINK=1 -DBOOST_SYSTEM_STATIC_LINK=1 -DNDEBUG  -I"." -c
-o "bin.v2/libs/fiber/build/gcc-mips/release/abi-o32/link-static/threading-multi/algo/work_stealing.
o" "libs/fiber/src/algo/work_stealing.cpp"

...failed gcc.compile.c++ bin.v2/libs/fiber/build/gcc-mips/release/abi-o32/link-static/threading-mul
ti/algo/work_stealing.o...
gcc.compile.c++ bin.v2/libs/fiber/build/gcc-mips/release/abi-o32/link-static/threading-multi/barrier
.o
{standard input}: Assembler messages:
{standard input}:1072: Error: opcode not supported on this processor: mips32 (mips32) `pause'
{standard input}:1154: Error: opcode not supported on this processor: mips32 (mips32) `pause'

```

ClaymorePT, 2018-09-11:
I'll check that out ASAP. I may have to disable fiber for that target.

thess, 2018-09-13:
FYI - Buildroot had same problem and fixed it by disallowing fiber for mips32/64 cpu types.
https://github.com/buildroot/buildroot/commit/914bd57db76941f4ca7c33eb8e217f06f790a6c2

ClaymorePT, 2018-09-13:
Yes, I might have to do the same. Fiber depends upon Boost.context which does not support mips32/64. Still, I want to do some tests if I only need to disable Fiber or if I have to disable Context instead.

thess, 2018-09-13:
Boost.context seemed to build OK if I built --without-fiber

ClaymorePT, 2018-09-13:
:+1:  Thank you for the hint. That makes my life easier. I will open a pull request asap with the fix.

ClaymorePT, 2018-09-14:
@thess I'm trying to mimic the Buildroot solution, but OpenWRT build system does not have a MIPS32 config variable. Is there a way for me to easily mimic the `BR2_MIPS_CPU_MIPS32` behaviour? 

I tried to use the `CPU_TYPE` variable and check if it was equal to `mips32` or `mips64` string but without success. 

The only alternative I can think of, is to disable Boost.Fiber for every target which uses these architectures.

ClaymorePT, 2018-09-18:
@thess Please check pull request https://github.com/openwrt/packages/pull/7067

thess, 2018-09-18:
Fixed via #7067

thess, 2018-09-20:
@ClaymorePT 
We still have a problem with the 'all libs' package when Boost.Fiber is not selected. The package build fails the dependency check at the end of the packaging and boost packages are still missing. Specifically, boost-fiber is still in the $(BOOST_DEPENDS) list which is referenced by Package/boost-libs. Short of removing the all-libs option, I haven't come up with any clever way of doing the package dependencies dynamically.

Any suggestions would help.

ClaymorePT, 2018-09-20:
@thess I just saw your pull request to solve this issue. If it solves the issue, for me its ok. 

The idea behind the `all libs` _package_ option was to provide an easy way to select all packages without having to go through every single one of them on the list.

What I don't understand is why the dependency check fails, when Fiber is clearly restricted to architectures not equal to mips32 and mips64. 

I have to be honest. KBuild to me is sometimes very cryptic. For example, there is no way to tell the user, when a packet cannot be selected due to the lack of a dependency. I give as an example, boost-python, boost-python3 and boost-locale. If these packets dependencies are not met, they are simply invisible in the package list. How can a user know the packages exist if they don't see it? 
It's not very user-friendly in my opinion. 

Once again, thank you for your support.

thess, 2018-09-20:
Well... There is a Kconfig option (not in the help) that by pressing <Z>, all hidden options with prompts defined are displayed but are not selectable "```- -```". You can then use \<Help\> to see the dependency.

PR #7081 (2018-09-20): boost: Fix mips32r1 cpu builds (exclude Boost.fiber)
Maintainer: @ClaymorePT 
Compile tested: brcm63xx, brcm47xx, ar71xx
Run tested: N/A

Description:
Exclude Boost.fiber when CPU_TYPE is either mips32 or mips64.

Found a way to leverage Kconfig syntax to create an invisible config variable suitable for testing in a package's DEPENDS string.

Closes #7000 (again)
closed 2018-09-20 by thess