openwrt/packages #7001: nut: Add missing parameters for the options available to the driver (usbhid-ups)

malakingpusa, 2018-09-12:
Maintainer: @cshoredaniel
Environment: (ar71xx, TP-Link AC1750 Archer C7 v4, OpenWrt 18.06.0 master branch)

Description:

This comes from results of testing pull request https://github.com/openwrt/packages/pull/6897.

Add missing parameters e.g. pollinterval, ignorelb, overrride.xxx to the options available to the driver.

The **/etc/config/nut_server** says:
```
config driver 'CP37'
	option driver usbhid-ups
	option port auto
        option vendorid 0764
        option productid 0501
        option serial CRNGV2000016 
        option pollinterval 15
        option offdelay 60
        option ignorelb 
```
After **/etc/init.d/nut-server** completes, the **/etc/nut/ups.conf** becomes:
```
root@TPLINKAC1750:~# cat /etc/nut/ups.conf 
# Config file automatically generated from UCI config
[CP37]
driver = usbhid-ups
port = auto
serial = CRNGV2000016
offdelay = 60
ondelay = 30
pollfreq = 30
vendorid = 0764
productid = 0501
```

cshoredaniel, 2018-09-12:
FYI the UCI for what I'm working in is here (not yet available, but soon): https://openwrt.org/docs/guide-user/services/ups/software.nut#nut_server

tecufanujacu, 2021-01-29:
Hello @danielfdickinson , sorry for the necropost but I think this is the better place to write.

I'm trying to use the override configs but without success, I need an ups.con as this:

[ups]
driver = blazer_usb
vendorid = xxxx
productid = xxxx
port = auto
override.battery.charge.low = 40

and I configured my nut_server in this way:
config driver 'ups'
        option driver blazer_usb
        option port auto
        option vendorid xxxx
        option productid xxxx
        option override override_battery_charge_low

config override 'override_battery_charge_low'
        option value 40

but it generate this ups.conf:
[nilox]
driver = blazer_usb
port = auto
vendorid = xxxx
productid = xxxx

I think that I'm doing some error, could you tell me how to correctly use the override configs?

danielfdickinson, 2021-02-06:
Sorry I shouldn't be listed as maintainer anymore for life reasons I am not active with OpenWrt anymore. Are you using openwrt/packages **master**? Another contributor found an error in the translation of UCI configuration to NUT configuration shell script, but it's probably not in 19.07.x.

tecufanujacu, 2021-02-06:
Hello, thanks for answering me, anyway I solved my problem. I was using a wrong configuration, "option override override_battery_charge_low" instead of "list override override_battery_charge_low".

doushang, 2022-12-12:
> Hello, thanks for answering me, anyway I solved my problem. I was using a wrong configuration, "option override override_battery_charge_low" instead of "list override battery_charge_low".

Hello,I'm trying to use the default configs but without success,  I configured my nut_server in this way:
```
config driver 'box850'
        option port 'auto'
        option driver 'usbhid-ups'
        list default 'input_voltage'

config default 'default_input_voltage'
        option value '150'
```
 could you tell me how to correctly use the default configs?