openwrt/packages #7012: Segmentation fault with prosody/liblua (Entware)

Rooby69, 2018-09-13:
Description:

Hello I got th info fom entware that I should contact you 
https://github.com/Entware/Entware/issues/107

I have issues with prosody(v0.9.12) and the liblua (v5.1.5) in entware on ASUS router RT-AC88U.
The prosody daemon crashes. gdb log:

Program received signal SIGSEGV, Segmentation fault.
0x2abf09a0 in luaL_addlstring () from /opt/lib/liblua.so.5.1.5
(gdb) bt
#0  0x2abf09a0 in luaL_addlstring () from /opt/lib/liblua.so.5.1.5
#1  0x2abefef8 in ?? () from /opt/lib/liblua.so.5.1.5
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

Can you pleas update liblua and prosody. 
Prosody has already a new scurity release 0.10.2 see
http://blog.prosody.im/prosody-0-10-2-security-release/

Thanks

dibdot, 2018-09-14:
Please always inform the package maintainer: @heil 
Also please try to reproduce your issue with a current/supported OpenWrt release - with logs, dumps etc.

Rooby69, 2018-09-15:
Unfrtunately I do not have anyadditional logs, dumps etc. I' not a linux expert.
I'm running prosody which uses liblua an cashes with the segmentatio fault shown above.
Router is a RT-AC88U running Asus Merlin SW with Entware.

Rooby69, 2018-09-16:
Anoher example:

Program received signal SIGSEGV, Segmentation fault.
0x2ab739a0 in luaL_addlstring (B=B@entry=0x7e9462e4, s=0x1994000 <error: Cannot access memory at address 0x1994000>, l=<optimized out>) at lauxlib.c:448
448     lauxlib.c: No such file or directory.
(gdb) bt
#0  0x2ab739a0 in luaL_addlstring (B=B@entry=0x7e9462e4, s=0x1994000 <error: Cannot access memory at address 0x1994000>, l=<optimized out>) at lauxlib.c:448
#1  0x2aea0730 in recvall (b=<optimized out>, buf=<optimized out>) at buffer.c:215
#2  buffer_meth_receive (L=0x2, buf=0x22b2a8) at buffer.c:125
#3  0x2ab672bc in luaD_precall (L=L@entry=0x23008, func=<optimized out>, func@entry=0x139790, nresults=nresults@entry=3) at ldo.c:320
#4  0x2ab71c24 in luaV_execute (L=L@entry=0x23008, nexeccalls=2254384, nexeccalls@entry=1) at lvm.c:847
#5  0x2ab6773c in luaD_call (L=0x23008, func=0x1396c0, nResults=-1) at ldo.c:378
#6  0x2ab66b30 in luaD_rawrunprotected (L=0x2ab66b30 <luaD_rawrunprotected+76>, L@entry=0x23008, f=0x23008, ud=0x23008, ud@entry=0x7e9485a8) at ldo.c:116
#7  0x2ab67898 in luaD_pcall (L=L@entry=0x23008, func=<optimized out>, u=u@entry=0x7e9485a8, old_top=224, ef=208) at ldo.c:464
#8  0x2ab63368 in lua_pcall (L=L@entry=0x23008, nargs=nargs@entry=0, nresults=nresults@entry=-1, errfunc=errfunc@entry=1) at lapi.c:881
#9  0x2ab74348 in luaB_xpcall (L=0x23008) at lbaselib.c:396
#10 0x2ab672bc in luaD_precall (L=L@entry=0x23008, func=<optimized out>, func@entry=0x1396a0, nresults=nresults@entry=-1) at ldo.c:320
#11 0x2ab71c24 in luaV_execute (L=L@entry=0x23008, nexeccalls=1063216, nexeccalls@entry=1) at lvm.c:847
#12 0x2ab6773c in luaD_call (L=0x23008, func=0x23298, nResults=-1) at ldo.c:378
#13 0x2ab66b30 in luaD_rawrunprotected (L=0x2ab66b30 <luaD_rawrunprotected+76>, L@entry=0x23008, f=0x0, ud=0x23008, ud@entry=0x7e948860) at ldo.c:116
#14 0x2ab67898 in luaD_pcall (L=L@entry=0x23008, func=<optimized out>, u=u@entry=0x7e948860, old_top=64, ef=48) at ldo.c:464
#15 0x2ab63368 in lua_pcall (L=0x23008, nargs=0, nresults=-1, errfunc=<optimized out>) at lapi.c:881
#16 0x000111f8 in _start ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

diizzyy, 2019-01-31:
Entware != Openwrt
@thess @hnyman 
Close as invalid

closed 2019-01-31 by hnyman