openwrt/packages #7016: cups: missing from newer repositories

anarcat, 2018-09-13:
Maintainer: @psycho-nico?
Environment: OpenWRT 18.06

Description:

Cups is missing from latest builds of OpenWRT or LEDE. It seems the package was not migrated from the [old packages repository](https://git.archive.openwrt.org/?p=packages.git;a=tree;f=net/cups;hb=HEAD). The [OpenWRT cups documentation](https://openwrt.org/docs/guide-user/services/print_server/cups.server) refers to a [third-party repository](https://github.com/Gr4ffy/lede-cups/) for "Installation on LEDE firmware" which I find confusing because it's unclear which firmware is LEDE and which is OpenWRT nowadays.

Multiple pull requests and issues were opened here to try and fix this (https://github.com/openwrt/packages/pull/1452, https://github.com/openwrt/packages/pull/1453, https://github.com/openwrt/packages/pull/1454, https://github.com/openwrt/packages/issues/2407) and in all of them @psycho-nico intervene to say they were working on cups already, in 2016. I guess nothing happened there so It would be nice to see cups come back in OpenWRT!

I have opened https://github.com/Gr4ffy/lede-cups/issues/6 to ask if the maintainer would be interested in merging their work here.

Hopefully this is according to process, apologies in advance for any possible misstep.

diizzyy, 2018-09-14:
If you want cups to be included again it needs to the current version or at least very close to it including dependencies.

dibdot, 2018-09-14:
@anarcat if you need this kind of functionality on your router, feel free to take over the the task to migrate such package from old repo and open an appropriate PR. Thanks.

closed 2018-09-14 by dibdot

anarcat, 2018-09-14:
well i'm happy to hear that people here are open to this being merged and I hope that @Gr4ffy will take this on. thanks for the feedback.