openwrt/packages #7042: mwan3: MWAN - Rules - Source address & squid

lucize, 2018-09-16:
@feckert adding multiple ip addresses to the `Source address` is possible but then the `iptables -vL -t mangle` will not show the rule in `Chain mwan3_rules`

also I'm trying to make squid work only on one interface (or on any)
if I add multiple source address in the rules (all ip on router as per issue stated above) then squid works but the tcp_outgoing_address will have no importance, testing on sites shows random random outgoing ip

squid works ok with interception but not if is manually added in the browser

when is manual set I get this error
`NONE/000 0 NONE error:transaction-end-before-headers - HIER_NONE/- -`
I opened a discussion at https://forum.openwrt.org/t/solved-mwan3-and-squid/20854

[image]

feckert, 2018-09-22:
> NONE/000 0 NONE error:transaction-end-before-headers - HIER_NONE/- -

What program generates this output?

lucize, 2018-09-22:
in squid access log, when you try multiple pages with manual proxy setup, no redirect (interception)

feckert, 2018-12-30:
@lucize no problem which version of mwan3 do you use on issue creation? The version 2.7.8 is the latest one from master.

lucize, 2018-12-30:
it was the one that had source route option

feckert, 2018-12-30:
It would be better if you could tell me the version ;-) or the openwrt version 17.01 /18.06 with this version information i know the mwan3 version

lucize, 2018-12-30:
I'm always on master,so I did an update to it (one per month or so), but I managed to satisfy my needs using transparent proxy because it was time pressing, I'll try to recreate the scenario with manual proxy setup

lucize, 2018-12-30:
I am pretty sure it was 2.7.2

feckert, 2019-03-04:
@lucize ping does your setup work with the last master?

neheb, 2020-01-12:
ping @lucize

lucize, 2020-01-12:
followed mwan commits and didn't see anything related to this, I'll close this as it seems to be a best effort support

closed 2020-01-12 by lucize