openwrt/packages #7077: mwan3: always shows INFO: MWAN not running

quazardous, 2018-09-19:
Maintainer: @Adze1502 
Environment: wrt1200ac / ARMv7 Processor rev 1 (v7l) / 18.06.1

Description: mwan3 seams to work (after global enabled, etc)
It detects wan / wan4g (via usb tether) switches (I'm using my laptop connected by wifi to a sub router of the main router with the ISP box shutdown ! nice ! )

but in lci mwan3 keeps telling me : INFO: MWAN not running

so I'm a little disapointed

so here is the Status > Load Balancing > Detail

MWAN Status - Detail
INFO: MWAN not running
```
Interface status:
 interface wan is offline and tracking is active
 interface wan4g is online and tracking is active

Current ipv4 policies:
balanced:
 wan4g (100%)

wan4g_only:
 wan4g (100%)

wan4g_wan:
 wan4g (100%)

wan_only:
 unreachable

wan_wan4g:
 wan4g (100%)

Current ipv6 policies:
balanced:
 default

wan4g_only:
 unreachable

wan4g_wan:
 unreachable

wan_only:
 unreachable

wan_wan4g:
 default

Directly connected ipv4 networks:
 127.0.0.0
 127.0.0.1
 192.168.2.1
 192.168.1.0
 192.168.42.21
 192.168.1.2
 192.168.2.255
 192.168.2.0/24
 192.168.42.255
 192.168.2.0
 192.168.42.0
 192.168.1.255
 192.168.1.0/24
 127.0.0.0/8
 192.168.42.0/24
 224.0.0.0/3
 127.255.255.255

Directly connected ipv6 networks:
 fd0f:68bd:4a21::/64
 fe80::/64

Active ipv4 user rules:
    0     0 - wan_wan4g  tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport sports 0:65535 multiport dports 22 
   91  8324 - wan_wan4g  all  --  *      *       192.168.2.101        0.0.0.0/0            
  671 47690 - balanced  all  --  *      *       0.0.0.0/0            0.0.0.0/0            

Active ipv6 user rules:
    0     0 - wan_wan4g  tcp      *      *       ::/0                 ::/0                 multiport sports 0:65535 multiport dports 22 
  100  7920 - balanced  all      *      *       ::/0                 ::/0
```

Here is my raw mwan conf:

```
config globals 'globals'
        option enabled '1'
        option mmx_mask '0x3F00'
        option local_source 'lan'

config interface 'wan'
	option enabled '1'
	list track_ip '8.8.4.4'
	list track_ip '8.8.8.8'
	list track_ip '208.67.222.222'
	list track_ip '208.67.220.220'
	option reliability '2'
	option count '1'
	option timeout '2'
	option interval '5'
	option up '3'
	option down '5'

config interface 'wan4g'
	list track_ip '8.8.8.8'
	list track_ip '208.67.220.220'
	option reliability '1'
	option count '1'
	option timeout '2'
	option interval '5'
	option enabled '1'
	option down '5'
	option up '3'

config member 'wan_m1_w3'
	option interface 'wan'
	option metric '1'
	option weight '3'

config member 'wan_m2_w3'
	option interface 'wan'
	option metric '2'
	option weight '3'

config member 'wan4g_m1_w2'
	option interface 'wan4g'
	option metric '1'
	option weight '2'

config member 'wan4g_m2_w2'
	option interface 'wan4g'
	option metric '2'
	option weight '2'

config policy 'wan_only'
	list use_member 'wan_m1_w3'

config policy 'wan4g_only'
	list use_member 'wan4g_m1_w2'

config policy 'balanced'
	list use_member 'wan_m1_w3'
	list use_member 'wan4g_m1_w2'
	option last_resort 'default'

config policy 'wan_wan4g'
	list use_member 'wan_m1_w3'
	list use_member 'wan4g_m2_w2'
	option last_resort 'default'

config policy 'wan4g_wan'
	list use_member 'wan_m2_w3'
	list use_member 'wan4g_m1_w2'

config rule 'ssh'
	option dest_port '22'
	option proto 'tcp'
	option use_policy 'wan_wan4g'
	option sticky '0'

config rule 'cloudy'
	option src_ip '192.168.2.101'
	option proto 'all'
	option sticky '0'
	option use_policy 'wan_wan4g'

config rule 'default_rule'
	option proto 'all'
	option sticky '0'
	option use_policy 'balanced'

```

here is my network config

```
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd0f:68bd:4a21::/48'

config interface 'lan'
	option type 'bridge'
	option ifname 'eth0'
	option proto 'static'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option ipaddr '192.168.2.1'

config interface 'wan'
	option ifname 'eth1'
	option _orig_ifname 'eth1'
	option _orig_bridge 'false'
	option proto 'static'
	option ipaddr '192.168.1.2'
	option netmask '255.255.255.0'
	option gateway '192.168.1.1'
	option metric '10'
	option dns '8.8.8.8 147.135.185.78'

config interface 'wan6'
	option ifname 'eth1'
	option proto 'dhcpv6'
	option reqaddress 'try'
	option reqprefix 'auto'
	option peerdns '0'
	option dns '2001:4860:4860::8888 2001:4860:4860::8844'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '0 1 2 3 5'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '4 6'

config interface 'wan4g'
	option proto 'dhcp'
	option ifname 'usb0'
	option peerdns '0'
	option metric '20'
	option dns '8.8.8.8 147.135.185.78'

```

dibdot, 2018-09-22:
inform current maintainer @feckert

quazardous, 2018-09-22:
@feckert wake up Neo !

feckert, 2018-09-22:
@quazardous 

> This is your last chance. After this, there is no turning back. You take the blue pill—the story ends, you wake up in your bed and believe whatever you want to believe. You take the red pill—you stay in Wonderland, and I show you how deep the rabbit hole goes. Remember: all I'm offering is the truth. Nothing more. 

I will take a look on this issue next week :-)

quazardous, 2018-09-28:
I had to do a rest + restore conf backup and the alert disappeared ...

closed 2018-09-28 by quazardous