openwrt/packages #7078: git: /usr/bin/git permissions set to 777

MikePetullo, 2018-09-20:
Maintainer: @tripolar 
Environment: x86_64 snapshot as of September 18, 2018 (https://downloads.openwrt.org/snapshots/packages/x86_64/packages/)

Description:
The git package appears to install /usr/bin/git with its permissions set to 777. Setting them to 755 seems more reasonable.

# ls -l git
ls: git: No such file or directory
# opkg install /tmp/git_2.19.0-1_x86_64.ipk 
Installing git (2.19.0-1) to root...
Configuring git.
# ls -l git
-rwxrwxrwx    2 root     root       2016380 Sep 18 11:12 git
# ls -l /usr/lib/git-core/git
-rwxrwxrwx    2 root     root       2016380 Sep 18 11:12 /usr/lib/git-core/git

tripolar, 2018-09-22:
fixed in https://github.com/openwrt/packages/commit/30898cc49d1fa9e73deb37ae82db5dc466941e1a

closed 2018-09-22 by tripolar

MikePetullo, 2018-09-23:
Confirmed fixed. Thank you!