openwrt/packages #7080: pci.ids cannot be downloaded

xback, 2018-09-20:
2 issues:
- The main download location has it's certificate expired
- All alts are not valid anymore.

Due to this, package pci-utils cannot be used anymore currently.

mkdir -p /mnt/ramdisk/koen/firmware/builds/generic_imx6/dl
SHELL= flock /mnt/ramdisk/koen/firmware/builds/generic_imx6/tmp/.pci.ids.4ad9327b147ce7bb6ae27b68efe0c1d89d9eb9a1.flock -c '  	/mnt/ramdisk/koen/firmware/builds/generic_imx6/scripts/download.pl "/mnt/ramdisk/koen/firmware/builds/generic_imx6/dl" "pci.ids.4ad9327b147ce7bb6ae27b68efe0c1d89d9eb9a1" "f8386c74ecc74f3c410b2f0f4885e7705c5dfcacd52fe924a931a63b246c8793" "pci.ids" "@GITHUB/pciutils/pciids/4ad9327b147ce7bb6ae27b68efe0c1d89d9eb9a1"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://raw.githubusercontent.com/pciutils/pciids/4ad9327b147ce7bb6ae27b68efe0c1d89d9eb9a1/pci.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://raw.githubusercontent.com/pciutils/pciids/4ad9327b147ce7bb6ae27b68efe0c1d89d9eb9a1/pci.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://raw.githubusercontent.com/pciutils/pciids/4ad9327b147ce7bb6ae27b68efe0c1d89d9eb9a1/pci.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://raw.githubusercontent.com/pciutils/pciids/4ad9327b147ce7bb6ae27b68efe0c1d89d9eb9a1/pci.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://raw.githubusercontent.com/pciutils/pciids/4ad9327b147ce7bb6ae27b68efe0c1d89d9eb9a1/pci.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.lede-project.org/pci.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/pci.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://downloads.openwrt.org/sources/pci.ids
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   184  100   184    0     0    184      0  0:00:01 --:--:--  0:00:01  1572
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
Makefile:48: recipe for target '/mnt/ramdisk/koen/firmware/builds/generic_imx6/dl/pci.ids.4ad9327b147ce7bb6ae27b68efe0c1d89d9eb9a1' failed
make[3]: *** [/mnt/ramdisk/koen/firmware/builds/generic_imx6/dl/pci.ids.4ad9327b147ce7bb6ae27b68efe0c1d89d9eb9a1] Error 2
make[3]: Leaving directory '/mnt/ramdisk/koen/firmware/builds/generic_imx6/feeds/packages/utils/pciutils'
time: package/feeds/packages/pciutils/compile#0.14#0.05#4.38
package/Makefile:107: recipe for target 'package/feeds/packages/pciutils/compile' failed
make[2]: *** [package/feeds/packages/pciutils/compile] Error 2
make[2]: Leaving directory '/mnt/ramdisk/koen/firmware/builds/generic_imx6'
package/Makefile:103: recipe for target '/mnt/ramdisk/koen/firmware/builds/generic_imx6/staging_dir/target-arm_cortex-a9+neon_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/mnt/ramdisk/koen/firmware/builds/generic_imx6/staging_dir/target-arm_cortex-a9+neon_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/ramdisk/koen/firmware/builds/generic_imx6'
/mnt/ramdisk/koen/firmware/builds/generic_imx6/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

diizzyy, 2018-09-21:
Seems to be a temporary local issue, URL works fine for me

dibdot, 2018-09-22:
As @diizzyy said, works for me as well.

closed 2018-09-22 by dibdot