openwrt/packages #7082: unbound-control - Error relocating /usr/sbin/unbound-control

s0m3f00l, 2018-09-21:
unbound-control 

Maintainer: @\<wcawijngaards>
Environment: TPLINK Archer C7 V2

Description:

Error when running unbound-control with default opkg install: 

```
root@LEDE_Edge:/usr/sbin# unbound-control --help
Error loading shared library libunbound.so.8: No such file or directory (needed by /usr/sbin/unbound-control)
Error relocating /usr/sbin/unbound-control: timehist_import: symbol not found
Error relocating /usr/sbin/unbound-control: log_err: symbol not found
Error relocating /usr/sbin/unbound-control: log_ident_set: symbol not found
Error relocating /usr/sbin/unbound-control: fname_after_chroot: symbol not found
Error relocating /usr/sbin/unbound-control: log_warn: symbol not found
Error relocating /usr/sbin/unbound-control: log_err_addr: symbol not found
Error relocating /usr/sbin/unbound-control: extstrtoaddr: symbol not found
Error relocating /usr/sbin/unbound-control: timehist_setup: symbol not found
Error relocating /usr/sbin/unbound-control: timehist_delete: symbol not found
Error relocating /usr/sbin/unbound-control: config_delete: symbol not found
Error relocating /usr/sbin/unbound-control: addr_is_ip6: symbol not found
Error relocating /usr/sbin/unbound-control: config_read: symbol not found
Error relocating /usr/sbin/unbound-control: options_remote_is_address: symbol not found
Error relocating /usr/sbin/unbound-control: fatal_exit: symbol not found
Error relocating /usr/sbin/unbound-control: config_create: symbol not found
Error relocating /usr/sbin/unbound-control: log_init: symbol not found
Error relocating /usr/sbin/unbound-control: ipstrtoaddr: symbol not found
```

EricLuehrsen, 2018-09-22:
What did you do now, exactly? `unbount-control` depends on `unbound` depends on `libunbound`. `/usr/lib/libunbound` should be visible after default install. Your problem is only this `Error loading shared library libunbound.so.8` and the rest are symptoms.

EricLuehrsen, 2018-09-22:
What did you do now, exactly? `unbount-control` depends on `unbound` depends on `libunbound`. `/usr/lib/libunbound` should be visible after default install. Your problem is only this `Error loading shared library libunbound.so.8` and the rest are symptoms. Also, NLNetlabs <wcawijngaards> is not responsible for unique distribution packages. A distribution has its own maintainers and in this case _me_.

EricLuehrsen, 2018-09-22:
What did you do now, exactly? `unbount-control` depends on `unbound` depends on `libunbound`. `/usr/lib/libunbound` should be visible after default install. Your problem is only this `Error loading shared library libunbound.so.8` and the rest are symptoms. Also, NLNetlabs \<wcawijngaards\> is not responsible for unique distribution packages. A distribution has its own maintainers and in this case me.

s0m3f00l, 2018-09-22:
Hey Man,

  Sorry, Not sure. But was getting that error with my openwrt distro. I thought I was missing libunbound.so.8 so I reinstalled my openwrt and then unbound,  from scratch. Now I'm getting this

root@OpenWrt:~# unbound
Error relocating /usr/lib/libunbound.so.8: getentropy: symbol not found
Error relocating /usr/sbin/unbound: getentropy: symbol not found
root@OpenWrt:~# 

of course unbound-control isnt working either.  And the /etc/init.d/unbound start just boot loops and crashes...  

This was the distro I was using 

https://forum.openwrt.org/t/ath79-builds-with-all-kmod-packages-through-opkg-flow-offloading/15897

Sorry for the trouble.

dibdot, 2018-09-22:
Sounds like a borked custom build - please discuss the follow up in the forum.

closed 2018-09-22 by dibdot