openwrt/packages #7083: mosquitto package version 1.5 failed to make for OpenWRT 15.05

congngale, 2018-09-21:
I have a problem when trying to build mosquitto package version 1.5 for OpenWRT 15.05. 

**The error as below:**

```
memory_mosq.o: In function `mosquitto__calloc':
memory_mosq.c:(.text+0x4c): undefined reference to `malloc_usable_size'
memory_mosq.o: In function `mosquitto(float, long double,...)(...)':
memory_mosq.c:(.text+0x88): undefined reference to `malloc_usable_size'
memory_mosq.o: In function `mosquitto__malloc':
memory_mosq.c:(.text+0xe0): undefined reference to `malloc_usable_size'
memory_mosq.o: In function `mosquitto__realloc':
memory_mosq.c:(.text+0x16c): undefined reference to `malloc_usable_size'
memory_mosq.c:(.text+0x190): undefined reference to `malloc_usable_size'
memory_mosq.o:memory_mosq.c:(.text+0x200): more undefined references to `malloc_usable_size' follow
collect2: error: ld returned 1 exit status
Makefile:59: recipe for target 'mosquitto' failed
```

The mosquitto package version 1.4.15 is no problem, but I want to use the latest version.

Thanks,
Cong Nga Le

karlp, 2018-09-21:
15.05 is unsupported.

closed 2018-09-21 by karlp

paldier, 2019-05-02:
ifeq ($(CONFIG_USE_UCLIBC),y)
	MAKE_FLAGS += WITH_MEMORY_TRACKING=no
endif

rhyttr, 2019-09-12:
> ifeq ($(CONFIG_USE_UCLIBC),y)
> MAKE_FLAGS += WITH_MEMORY_TRACKING=no
> endif

This works!

tirzasrwn, 2022-03-10:
> ifeq ($(CONFIG_USE_UCLIBC),y) MAKE_FLAGS += WITH_MEMORY_TRACKING=no endif

Thanks, this also work for me.