openwrt/packages #7096: nut: Broken blazer_usb after PR #6897

ivanich, 2018-09-24:
Maintainer: @cshoredaniel
Environment: (ath79, TP-Link wdr4300, OpenWrt 18.06.0 master branch)
Description:
Using blazer_usb nut driver and found that this PR brings https://github.com/openwrt/packages/pull/6897/files#diff-b43121d3cb6f04f9a4f58f93c7970813R150 option which is not available in blazer_usb thus it cannot start with the following error
Wed Sep 19 08:56:12 2018 daemon.info blazer_usb[3871]: Network UPS Tools - Megatec/Q1 protocol USB driver 0.12 (reboot-8069-ge8cbfedc72)
Wed Sep 19 08:56:12 2018 daemon.info blazer_usb[3871]:
Wed Sep 19 08:56:12 2018 daemon.info blazer_usb[3871]: Fatal error: 'pollfreq' is not a valid variable name for this driver.
Wed Sep 19 08:56:12 2018 daemon.info blazer_usb[3871]:

The second issue is permissions to /dev/bus/usb/001/00*
I can get it working with 2 steps:

    commenting out pollfreq from /etc/init.d/nut-server
    adding rw perms to /dev/bus/usb/001/00*

I'm using latest master compiled from sources
P.S. Not sure that second issue with permissions is related to PR #6897 .

cshoredaniel, 2018-10-09:
Permissions issue may be related to https://github.com/openwrt/packages/issues/6966.  Otherwise it's a missing productid/vendorid in NUT upstream I'd think.  Other issue I am working on (the invalid variable).

mantinan, 2018-10-19:
Hi!
I have downloaded master from cshoredaniel:pr-nut-more-fixes and compiled it.

I'm finding that even though I have set it to run as nut I get upsd running as root, this used to be user nut on -8 version, so -9 has a regression here.

On -8 a ps would tell:
 1172 nut       2900 S    /usr/sbin/upsd -D -u nut
On -9 I get:
 1258 root      2896 S    /usr/sbin/upsd -D

This is my nut_server config:
config driver 'riello'
	option driver riello_usb
	option port /dev/ttyUSB0
config listen_address
	option address 172.22.54.49
	option port 3493
config upsd 'upsd'
	option maxage 15
	option statepath /var/run/nut
	option maxconn 1024
	option runas nut

mantinan, 2018-10-22:
Great!

cshoredaniel, 2018-12-11:
@ivanich FYI I've posted PR: https://github.com/openwrt/packages/pull/7638 which ought to do must of what's now in master except the USB hotplug bits (as that's a new feature IMO).  If you can, please test and comment!

cshoredaniel, 2018-12-17:
@ivanich Thank you for letting me know!