openwrt/packages #7108: mwan3 start problem and after ping 8.8.8.8 network unreachable

daidaiwood, 2018-09-27:
# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)
```      
openwrt  18.06.1
mwan3    2.6.17
```
Description:
```
   i use the command /etc/init.d/mwan3 start and got some problem.
   uci: Entry not found
   uci: Entry not found
   uci: Entry not found
   grep: /var/etc/miniupnd.conf: No such file or directory
   uci: Entry not found

   after execute this command,my internet network is unreachable.
   help me.
   thx.
![qq 20180927111045](https://user-images.githubusercontent.com/43629170/46122109-88e65e00-c249-11e8-9fec-29e834f13354.png)

```

daidaiwood, 2018-09-28:
After another version of the test,  found it conflicts with sfe.

dibdot, 2018-10-05:
Please always inform the maintainer, too: @feckert

feckert, 2018-10-09:
mwan3 does not use **/var/etc/miniupnpd.conf** so maybe the problem has todo with this package?

daidaiwood, 2018-10-12:
> mwan3 does not use **/var/etc/miniupnpd.conf** so maybe the problem has todo with this package?

/etc/hotplug.d/50-miniupnpd cause this problem.
move this file to other place,then run wman3,it's good.

closed 2018-10-12 by daidaiwood

ehrmann, 2020-06-21:
I ran into the problem, too. The `grep: /var/etc/miniupnd.conf: No such file or directory` error is benign, but I opened a PR to hide it. The underlying cause that that I had miniupnpd installed, but not enabled. Enabling it removed the error (or uninstalling miniupnpd). I wouldn't move the hotplug file because it will just confuse opkg later, or, worst case, cause hard-to-debug problems when you enable miniupnpd.