openwrt/packages #7119: ddns-scripts additional install ddns-scripts_cloudflare issues

SUNJINZE, 2018-09-29:
After compiling ddns-scripts, you need to install "ddns-scripts_cloudflare". Is there a way to add the compiled package? Avoid installing ddns-scripts_cloudflare every time.

dibdot, 2018-09-29:
No, if you use cloudfare you have to install this add-on as well.

closed 2018-09-29 by dibdot

SUNJINZE, 2018-09-29:
> No, if you use cloudfare you have to install this add-on as well.

Thanks for your answer, Can this add-on be included in the ddns script? Don't need to install ddns-scripts_cloudflare again

ttimasdf, 2018-12-21:
```
root@whatever:~# opkg info ddns-scripts_cloudflare.com-v4
Package: ddns-scripts_cloudflare.com-v4
Version: 2.7.8-1
Depends: ddns-scripts, curl
Status: install user installed
Section: net
Architecture: all
Size: 4042
Filename: ddns-scripts_cloudflare.com-v4_2.7.8-1_all.ipk
Description: Dynamic DNS Client scripts extension for CloudFlare.com API-v4 (require/install cURL)
Installed-Time: 1545190210
```
The curl dependency is the key.