openwrt/packages #7125: Seed GitHub PR's with recommended reviewers

pprindeville, 2018-09-30:
When starting a PR, the package maintainer isn't always listed as a potential reviewer.  Which leaves you wondering, "Okay, who's going to review my PR?"

Per @champtar, GitHub offers some functionality already when one uses a `/.github/CODEOWNERS` files.  Its format is documented here:

https://help.github.com/articles/about-codeowners/

We could synthesize this file based on the `MAINTAINER` from each project's `Makefile`, as well as the Authors and Committers from the git-log for that directory.

champtar, 2018-10-01:
Doesn't handle multi maintainer
```
git grep 'MAINTAINER:=.*@' | grep Makefile | sed 's#/Makefile.*<\([^>]*\)>.*$#/\t\1#' > .github/CODEOWNERS
```

champtar, 2018-10-01:
Also CODEOWNERS feature only warn people with commit access