openwrt/packages #7130: unbound or openvswitch: OpenSSL does not support ECDSA: please upgrade or rerun with --disable-ecdsa

froodproton, 2018-10-01:
Please make sure that the issue subject starts with unbound or openvswitch

this is the same as https://github.com/openwrt/packages/issues/6993

Building the latest master for ar71xx_generic_DEVICE_archer-c7-v2
with config from https://downloads.openwrt.org/releases/18.06.1/targets/ar71xx/generic/config.seed
fails with error
```
checking for ECDSA_sign... no
configure: error: OpenSSL does not support ECDSA: please upgrade or rerun with --disable-ecdsa
Makefile:193: recipe for target '/home/frood/Documents/git/OwrtLEDE/openwrt/build_dir/target-mips_24kc_musl/unbound-1.8.0/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/home/frood/Documents/git/OwrtLEDE/openwrt/build_dir/target-mips_24kc_musl/unbound-1.8.0/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/frood/Documents/git/OwrtLEDE/openwrt/feeds/packages/net/unbound'
```

dibdot, 2018-10-05:
Why didn't you inform the openvswitch maintainer? Mentioned in the referenced ticket and the issue template. Without that you will only produce stale tickets ...

yousong, 2018-10-10:
I am afraid this is an unbound issue ;) let's move further discussion there at #6993

closed 2018-10-10 by yousong

EricLuehrsen, 2018-10-12:
How is this a Unbound problem? ECDSA is a supported algorithm for DNSSEC and TLS. DSKEY records label it `13` in the algo field. If you compile OpenSSL without it, then Unbound **should** complain the library is short of its requirements. Worth reading: https://www.cloudflare.com/dns/dnssec/ecdsa-and-dnssec/