openwrt/packages #7135: ddns-scripts_no-ip_com not updating

kurac338, 2018-10-03:
Hi, I am having trouble setting up the No-IP.com DDNS on my TP-Link Archer C7.

I am using the Luci web interface to configure the client. My router sits behind a modem, so I have it set up to retrieve the public IP from an URL, which works fine.

The problem is that the newly leased public IP never gets reflected on the provider's side.

The script detects a change in the IP address, and tries to update it. Sometime I get an error at that point, other times it says the (force) update completed successfully. BUT the problem is that in both cases I am still registered on the providers side with my old public IP. Its as if the request does not complete successfully, and that the log is incorrect.

I am currently at work, so I don't have any access to configuration files or logs. Please describe what files are needed to debug this situation, and I will gladly provide them :)

Thanks.

kofec, 2018-10-03:
There is an issue that you have to install real "wget" as tiny substitution is not working for no-ip.

kofec, 2018-10-03:
Not it is linked to uclient-fetch
/bin/wget -> uclient-fetch

kurac338, 2018-10-04:
My wget is located at /usr/bin/wget and points to wget-ssl package, so I think this is configured correctly.

kurac338, 2018-10-04:
I've diagnosed the issue, the error was on my end. I misunderstood the configuration parameters.

closed 2018-10-04 by kurac338