openwrt/packages #7162: openssh-sftp-server: compilation fails without full version of openssh server

zokl, 2018-10-09:
Maintainer: @cotequeiroz
Environment:  all, openssh, OpenWrt 18.06.1

Description:

If I use only openssh-sftp-server the compilation fails on this point:

```
_SOURCE -DHAVE_CONFIG_H -c kludge-fd_set.c mipsel-openwrt-linux-musl-gcc -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/develict/data/deveplatform_v2/build_dir/target-mipsel_24kc_musl/openssh-without-pam/openssh-7.8p1:openssh-7.8p1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -pipe -Wall -Wpointer-arith -Wuninitialized -Wsign-compare -Wformat-security -Wsizeof-pointer-memaccess -Wno-pointer-sign -Wno-unused-result -fno-strict-aliasing -ftrapv -fno-builtin-memset -fstack-protector-strong -fPIE -I. -I.. -I. -I./.. -I/home/develict/data/deveplatform_v2/staging_dir/target-mipsel_24kc_musl/usr/include -I/home/develict/data/deveplatform_v2/staging_dir/target-mipsel_24kc_musl/include -I/home/develict/data/deveplatform_v2/staging_dir/toolchain-mipsel_24kc_gcc-7.3.0_musl/usr/include -I/home/develict/data/deveplatform_v2/staging_dir/toolchain-mipsel_24kc_gcc-7.3.0_musl/include/fortify -I/home/develict/data/deveplatform_v2/staging_dir/toolchain-mipsel_24kc_gcc-7.3.0_musl/include -D_XOPEN_SOURCE=600 -D_BSD_SOURCE -D_DEFAULT_SOURCE -DHAVE_CONFIG_H -c openssl-compat.c In file included from openssl-compat.c:32:0: openssl-compat.h:27:10: fatal error: openssl/ecdsa.h: No such file or directory #include <openssl/ecdsa.h> ^~~~~~~~~~~~~~~~~ compilation terminated.
```
If openssh-sftp-server is compiled with the openssh-server package all goes well.

cotequeiroz, 2018-10-09:
I will take a look at it, as I probably introduced the bug, but to be proper, package maintainer is @tripolar.

tripolar, 2018-10-09:
@cotequeiroz if you can't find a solution i will drop your openssl patch, because the next version will have the openssl patch in mainline and tested. Somehow i expected bugs like this

cotequeiroz, 2018-10-09:
Here's the PR I've opened upstream:
https://github.com/openssh/openssh-portable/pull/101
I'm opening one here as well.

zokl, 2018-10-09:
Thank you very much for your help.

cotequeiroz, 2018-10-19:
Openssh 7.9p1 was released, but they didn't address my bug report.  Your problem in particular might have been fixed if you build without openssl at all.  However, if openssl is selected and ECC is turned off, the problem remains.  I've updated the pull request to use 7.9p1.

cotequeiroz, 2018-10-19:
I've run some more tests, and 7.9p1 does not fix this.  I now understand why it is failing: If you select openssh-sftp-server, then openssl does not get selected as a dependency, but it is built anyway (I don't understand why it is getting built).  Since it was not selected in .config, no options are set either, not even the default ones.  So, a bare openssl library gets built, not including ECC support.
Then, when openssh-sftp-server gets built, it fails because of the bug I reported.
If you select the openssh-server package, then libopenssl is pulled as a dependency, and its default options get applied, including ECC, so the bug is not triggered.

cotequeiroz, 2018-10-24:
This was fixed with 08c5f7cfabf6f01890b11905dd12661d7a33a9b6, so this issue should be closed.

zokl, 2018-10-24:
> This was fixed with 08c5f7c, so this issue should be closed.

Thank you!

S pozdravem,
Zbyněk Kocur
-- 
email: zbynek.kocur@fel.cvut.cz
phone: +420 22435 4054
web: http://www.fel.cvut.cz
Department of Telecommunications Engineering
Faculty of Electrical Engineering
Czech Technical University in Prague
Technicka 2
166 27 Praha 6

closed 2018-10-24 by tripolar