openwrt/packages #7168: <php7>: some problem with zlib

tony-sunjian, 2018-10-10:
Please make sure that the issue subject starts with `<package-name>: `

This repo here is only for packages maintained in this repo.  For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses

 - OpenWrt base system: https://bugs.openwrt.org
 - Most LuCI packages: https://github.com/openwrt/luci/issues

# Issue template (remove lines from top till here)

Maintainer: @mhei 
Environment:
Model: Thunder Timecloud
Architecture: MediaTek MT7621 ver:1 eco:3
Firmware Version: OpenWrt 18.06.1 r7258-5eb055306f / LuCI openwrt-18.06 branch (git-18.228.31946-f64b152)
Kernel Version 4.14.63

Description:
I have installed next packages: 
`php7 php7-cli php7-mod-zip php7-mod-fastcgi php7-mod-fpm php7-mod-openssl php7-mod-ctype`

When I run a test script, it show me next:

`
        $url = "http://app.zblogcn.com/client/";
        $ch = curl_init($url);
	if (extension_loaded('zlib')) {
		curl_setopt($ch, CURLOPT_ENCODING, 'gzip');
	}
	curl_setopt($ch, CURLOPT_RETURNTRANSFER, 1);
	curl_setopt($ch, CURLOPT_TIMEOUT, 120);
	if(isset($_SERVER['HTTP_ACCEPT'])){
		curl_setopt($ch,CURLOPT_HTTPHEADER,array (
			 'Accept: ' . $_SERVER['HTTP_ACCEPT']
		));
	}
	curl_setopt($ch, CURLOPT_SSL_VERIFYPEER, 0);
	curl_setopt($ch, CURLOPT_SSL_VERIFYHOST, 0);
	if (ini_get("safe_mode") == false && ini_get("open_basedir") == false) {
		curl_setopt($ch, CURLOPT_MAXREDIRS, 10);
		curl_setopt($ch, CURLOPT_FOLLOWLOCATION, 1);
	}

	$r = curl_exec($ch);
	echo $r;
	echo curl_error($ch);
	curl_close($ch);`

when i use "curl_setopt($ch, CURLOPT_ENCODING, 'gzip');"  

php reports an exception :  "Unrecognized content encoding type. libcurl understands identity content encodings."

if i remove the "curl_setopt($ch, CURLOPT_ENCODING, 'gzip');"  php get the right result;

diizzyy, 2018-10-10:
(lib)curl doesn't support zlib encoding by default, you need to build your own package.

tony-sunjian, 2018-10-11:
thanks for reply @diizzyy @mhei