openwrt/packages #7171: <python>: Python 2.7.15 Darwin compile error

leonardpitzu, 2018-10-10:
Maintainer: @\>jan-kardell>
Environment: (ath79, archer c7 v2, OpenWrt Master)

Description: python 2.7.15 fails to compile on macos mojave

```
checking whether the C compiler works... no
configure: error: in `/Volumes/openwrt/openwrt/build_dir/hostpkg/Python-2.7.15':
configure: error: C compiler cannot create executables
```

rossbcan, 2018-10-11:
x86_64 also...

HOST_LDFLAGS += \
        $$$$(pkg-config --static --libs libcrypto libssl) -L$(STAGING_DIR_HOST)/lib -Wl$(comma)-rpath-link=$(STAGING_DIR_HOST)/lib

gets U past this, then to:
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... configure: error: in `/home/rossb/OpenWrt/x86_64/r8210/trunk/build_dir/hostpkg/Python-2.7.15':
configure: error: cannot run C compiled programs.
If you meant to cross compile, use `--host'.

checking config.log:
configure:4127: ./conftest
./conftest: error while loading shared libraries: libcrypto.so.1.0.0: cannot open shared object file: No such file or directory

So, now a problem with LD_LIBRARY_PATH, which I am stuck at (so far)

...Bill

rossbcan, 2018-10-11:
Adding:
define Host/Configure
        export LD_LIBRARY_PATH=$(LD_LIBRARY_PATH):$(STAGING_DIR_HOST)/lib; $(call Host/Configure/Default)
endef

gets you to:

./python -E -S -m sysconfig --generate-posix-vars ;\
if test $? -ne 0 ; then \
	echo "generate-posix-vars failed" ; \
	rm -f ./pybuilddir.txt ; \
	exit 1 ; \
fi
./python: error while loading shared libraries: libcrypto.so.1.0.0: cannot open shared object file: No such file or directory

so, its still a host library search problem

if you set (in your build shell) LD_LIBRARY_PATH to include $(STAGING_DIR_HOST)/lib, SUCCESS

I will leave it to the package maintainer to determine how best to sort this out.

...B

jefferyto, 2019-01-08:
@rossbcan it looks like you are getting a different error than OP, would you mind opening a new issue?

@commodo any thoughts on this?

commodo, 2019-01-09:
I'll take a look.
I missed this one.

cshoredaniel, 2019-01-09:
At least with the SDK I had similar 'random' issues due path names exceeding some limit in the toolchain (I haven't figured out where yet).  Can you try doing the compilation in e.g. /home/user/sdk (e.g. tar --strip-components=1 -C /home/user/sdk -xvJf path/to/sdk.tar.xz instead? (or if you don't use SDK just move the openwrt source dir to /home/user/openwrt?

commodo, 2019-02-26:
@leonardpitzu 
has this issue been resolved ?
or if you resolved it, how ?

I did not find a MacOS Mojave around ; I have some older MacOS version, on some Mac Mini that is shared by other people, so I can't test there without updating and I don't want to interfere with it [since it's shared]

I can try to create a fix, but I would need someone to test it [if you're willing to help].

Thanks
Alex

leonardpitzu, 2019-02-26:
@commodo issue still present (tested a few days ago). Drop a fix and i will test and report back.

leonardpitzu, 2019-03-04:
@commodo This solves the issue

ifeq ($(HOST_OS),Darwin)
HOST_LDFLAGS += \
	$$$$(pkg-config --static --libs libcrypto libssl)
else
HOST_LDFLAGS += \
	$$$$(pkg-config --static --libs libcrypto libssl) -Wl$(comma)-rpath=$(STAGING_DIR_HOSTPKG)/lib
endif

commodo, 2019-03-05:
@leonardpitzu 
Many thanks for providing the solution :)

I've not had time to allocate for this yet.

I will try to test this a bit on Linux.
I am curios if just removing ` -Wl$(comma)-rpath=$(STAGING_DIR_HOSTPKG)/lib` works for Linux as well, in which case there wouldn't be a need for the `ifeq ($(HOST_OS),Darwin)` check

Thanks
Alex

jan-kardell, 2019-03-05:
Alexandru Ardelean skrev:

> @leonardpitzu <https://github.com/leonardpitzu>
> Many thanks for providing the solution :)
>
> I've not had time to allocate for this yet.
>
> I will try to test this a bit on Linux.
> I am curios if just [...]
No it does not work, without it program uses system libraries instead of 
those installed in staging_dir. It just happens to work on some specific 
version of Ubuntu, but not other versions and distros.

//Jan

commodo, 2019-03-07:
@leonardpitzu 
on another note
can you try 2.7.16 ?

https://github.com/commodo/packages/commits/python-2-7-16
https://github.com/commodo/packages/commit/fcc016b7309e5d4f28731ef71fcbc157a7e3b66f

I'm curios if there's a fix in there.
Looking at the Misc/NEWS file I can't tell if there's anything for sure.

leonardpitzu, 2019-03-09:
tested with a clean build right now and issue is still present. the patch is still needed.

commodo, 2019-03-11:
could you post the entire build log somewhere ?

thing is, the patch you suggested just tells the build to use the host's OpenSSL libraries, vs using OpenWrt's libressl libs;

if possible, i would like to fix Python's build with OpenWrt's libressl

leonardpitzu, 2019-03-20:
@commodo- i'll try a buiuld this evening and post a log afterwords. isn't mbedtls the "default" ssl libn in openwrt? i thought libressl is used only for the toolchain. lucissl is also using mbedtls as far as i know (may be wrong though...)

commodo, 2019-03-20:
i am not sure about `mbedtls` ; i think it was the default ssl lib in openwrt at some point ; i did not follow if anything changed;
libressl in `tools/libressl` is for the host-build tools;

PR #9762 (2019-08-17): [openwrt-19.07] python,python3: Fix host Python compilation for macOS
Maintainer: me, @commodo 
Compile tested: none (cherry-picked from #9760)
Run tested: none

Description:
* Remove `$$$$(pkg-config --static --libs libcrypto libssl)` from `HOST_LDFLAGS`

  Having this leads to an `unknown type name 'u_int'` error on Mac. Removing it doesn't appear to affect Python's ability to find buildroot LibreSSL.

* Change `-Wl,-rpath=...` to `-Wl,-rpath,...` in `HOST_LDFLAGS`

  The equals sign version is not supported by the Mac linker (appears to be an GNU extension). The comma version is supported;  [... cut]
commit e4507482 (2019-08-17): python,python3: Fix host Python compilation for macOS

* Remove $$$$(pkg-config --static --libs libcrypto libssl) from
  HOST_LDFLAGS

  Having this leads to an "unknown type name 'u_int'" error on Mac.
  Removing it doesn't appear to affect Python's ability to find
  buildroot LibreSSL.

* Change -Wl,-rpath=... to -Wl,-rpath,... in HOST_LDFLAGS

  The equals sign version is not supported by the Mac linker (appears to
  be an GNU extension). The comma version is supported; -rpath and its
  argument will be separated by a space when passed to the linker.

* Add ac_cv_header_libintl_h=no to HOST [... cut]
PR #9760 (2019-08-17): python,python3: Fix host Python compilation for macOS
Maintainer: me, @commodo 
Compile tested: armvirt-64, 2019-08-16 snapshot sdk (Linux) / x86-64, OpenWrt master (Mac)
Run tested: manual testing with host Python (`import ssl`, `ssl.OPENSSL_VERSION`), checked rpath

Description:
* Remove `$$$$(pkg-config --static --libs libcrypto libssl)` from `HOST_LDFLAGS`

  Having this leads to an `unknown type name 'u_int'` error on Mac. Removing it doesn't appear to affect Python's ability to find buildroot LibreSSL.

* Change `-Wl,-rpath=...` to `-Wl,-rpath,...` in `HOST_LDFLAGS`

  The equals sign ve [... cut]
commit 229c23a4 (2019-08-17): python,python3: Fix host Python compilation for macOS

* Remove $$$$(pkg-config --static --libs libcrypto libssl) from
  HOST_LDFLAGS

  Having this leads to an "unknown type name 'u_int'" error on Mac.
  Removing it doesn't appear to affect Python's ability to find
  buildroot LibreSSL.

* Change -Wl,-rpath=... to -Wl,-rpath,... in HOST_LDFLAGS

  The equals sign version is not supported by the Mac linker (appears to
  be an GNU extension). The comma version is supported; -rpath and its
  argument will be separated by a space when passed to the linker.

* Add ac_cv_header_libintl_h=no to HOST [... cut]
closed 2019-08-17 by neheb