openwrt/packages #7180: mc: Buildbot is failing due to missing librpc dependency

neheb, 2018-10-11:
@dibdot 

https://downloads.openwrt.org/snapshots/faillogs/mipsel_24kc/packages/mc/compile.txt

jow-, 2018-10-12:
@dibdot  @neheb - try `CONFIGURE_VARS += ac_cv_func_pmap_set=no`

dibdot, 2018-10-12:
@jow- many thanks, I'll try that.

dibdot, 2018-10-12:
@jow- I've added this variable to my Makefile but mc generation dies here locally as long as I build librpc in parallel.

tripolar, 2018-10-13:
librpc should get dropped/removed. See https://github.com/openwrt/openwrt/pull/1280

diizzyy, 2018-11-17:
Close this issue as it's fixed

dibdot, 2018-11-17:
@diizzyy I would like to keep this open as long as the underlying librpc issue hasn't be fixed.

lyssdod, 2019-01-19:
Hi @neheb and @dibdot, are there any news regarding this? The buildlog you've provided is 404 now

neheb, 2019-01-19:
This has been fixed AFAIK

dibdot, 2019-01-19:
Well, not really fixed - I've just disabled VFS support by default (again). Anyway, let's close this one for now.

closed 2019-01-19 by dibdot

lyssdod, 2019-01-19:
@dibdot but what needs to be done to fix it properly? Fix build with libtrpc?