openwrt/packages #7195: <ddns-scripts>: incorrectly interprets response from namecheap

zxdavb, 2018-10-13:
Firstly thanks for a great package!

Maintainer: @dibdot 
Environment:  OpenWrt 18.06-SNAPSHOT, r7317-f506de2cda

Description: When chasing a bug of my own devising (the `192.168.0.100`, below), I found this in the logs:
```
Sat Oct 13 10:11:19 2018 user.warn ddns-scripts[9587]: namecheap: Updating IP at DDNS provider failed - starting retry 96/0
```

Part of the issue is caused by `ddns-scripts` incorrectly interpreting the response from **namecheap** as successful, when it was not, due to an incorrect password (below is anonymised):
```
 101119       : Detect registered/public IP
 101119       : #> /usr/bin/host -t A -4 domain.co.uk >/var/run/ddns/namecheap.dat 2>/var/run/ddns/namecheap.err
 101119       : Registered IP '192.168.0.100' detected
 101119  WARN : Updating IP at DDNS provider failed - starting retry 96/0
 101119       : Detect local IP on 'web'
 101120       : #> /usr/bin/wget-ssl -nv -t 1 -O /var/run/ddns/namecheap.dat -o /var/run/ddns/namecheap.err -4 --no-proxy 'http://myip.dnsomatic.com/'
 101120       : Local IP '87.62.58.193' detected on web at 'http://myip.dnsomatic.com/'
 101120       : Update needed - L: '87.62.58.193' <> R: '192.168.0.100'
 101121       : #> /usr/bin/wget-ssl -nv -t 1 -O /var/run/ddns/namecheap.dat -o /var/run/ddns/namecheap.err -4 --no-proxy 'http://dynamicdns.park-your-domain.com/update?host=%40&domain=domain.co.uk&password=***PW***&ip=87.62.58.193'
 101121       : DDNS Provider answered:
<?xml version="1.0"?><interface-response><Command>SETDNSHOST</Command><Language>eng</Language><ErrCount>1</ErrCount><errors><Err1>Passwords do not match</Err1></errors><ResponseCount>1</ResponseCount><responses><response><ResponseNumber>304156</ResponseNumber><ResponseString>Validation error; invalid ; password</ResponseString></response></responses><Done>true</Done><debug><![CDATA[]]></debug></interface-response>
 101122  info : Update successful - IP '87.62.58.193' send
 101122  info : Forced update successful - IP: '87.62.58.193' send
 101122       : Waiting 600 seconds (Check Interval)```

dibdot, 2018-10-14:
@zxdavb Nice try - I'm **not** maintainer of this epic script eldorado. Anyway I try to keep this running, but I'm not using this package ... you have been warned! ;-))

Looking at the code I think it works as designed. In "services" file the namecheap definition looks like this:

```
"namecheap.com"		"http://dynamicdns.park-your-domain.com/update?host=[USERNAME]&domain=[DOMAIN]&password=[PASSWORD]&ip=[IP]"
```
There is no third column with a possible update answer string defined.
In dynamic_dns_functions.sh whithin the send_update() function I found the following code snippet:
```
[...]
		write_log 7 "DDNS Provider answered:${N}$(cat $DATFILE)"

		[ -z "$UPD_ANSWER" ] && return 0	# not set then ignore
[...]
```
That means the ddns package will always return 0 as long as no UPD_ANSWER string has been defined. Feel free to define a valid answer string for namecheap and open a PR to this repo. Thanks!

zxdavb, 2018-10-15:
Ha! I meant to write @chris5560 & got confused (**adblock** is my other go-to package).

Fair enough about the PR - I will add this to my list!

For my own reference, in: https://www.namecheap.com/support/api/methods/domains-dns/set-custom.aspx, we have an example response:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<ApiResponse xmlns="http://api.namecheap.com/xml.response" Status="OK">
  <Errors />
  <RequestedCommand>namecheap.domains.dns.setCustom</RequestedCommand>
  <CommandResponse Type="namecheap.domains.dns.setCustom">
    <DomainDNSSetCustomResult Domain="domain.com" Updated="true" />
  </CommandResponse>
  <Server>SERVER-NAME</Server>
  <GMTTimeDifference>+5</GMTTimeDifference>
  <ExecutionTime>32.76</ExecutionTime>
</ApiResponse>
```

This is not in the same format as my error, above - but it's using a much different URL!

zxdavb, 2018-10-15:
Options to try would be 
a) use answer = `<ErrCount>0`, 
b) use api.namecheap.com rather than dynamicdns.park-your-domain.com, 
c) re-write scripts to handle answers of failure (otherwise assume OK), as well as answers of success (otherwise assume failed)

I will try a) & b) & report back...

dibdot, 2018-10-25:
@zxdavb any news on this?

zxdavb, 2018-10-25:
Option a) works great, although I am using: `"<ErrCount>0<\ErrCount>"` in the third column.

dibdot, 2018-10-30:
Great, thanks for your efforts. Since this a configuration issue, I will close this issue. Feel free to open a PR with that changed config.

closed 2018-10-30 by dibdot