openwrt/packages #7202: gcc: cannot compile GCC package to .ipk

805447391, 2018-10-16:
Env:
VirtualBox uBuntu 16.04.10
OpenWrt target: X86 normal

Log:
wang@wang-VirtualBox:~/Desktop/openwrt-x86$ make package/gcc/compile V=s
make[2]: Entering directory '/home/wang/Desktop/openwrt-x86/scripts/config'
make[2]: Leaving directory '/home/wang/Desktop/openwrt-x86/scripts/config'
make[1]: Entering directory '/home/wang/Desktop/openwrt-x86'
make[2]: Entering directory '/home/wang/Desktop/openwrt-x86/package/libs/toolchain'
echo "libc" >> /home/wang/Desktop/openwrt-x86/staging_dir/target-i386_pentium4_glibc/pkginfo/toolchain.default.install
echo "libgcc" >> /home/wang/Desktop/openwrt-x86/staging_dir/target-i386_pentium4_glibc/pkginfo/toolchain.default.install
echo "libssp" >> /home/wang/Desktop/openwrt-x86/staging_dir/target-i386_pentium4_glibc/pkginfo/toolchain.default.install
echo "libpthread" >> /home/wang/Desktop/openwrt-x86/staging_dir/target-i386_pentium4_glibc/pkginfo/toolchain.default.install
echo "librt" >> /home/wang/Desktop/openwrt-x86/staging_dir/target-i386_pentium4_glibc/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/wang/Desktop/openwrt-x86/package/libs/toolchain'
time: package/libs/toolchain/compile#0.17#0.05#3.24
make[2]: Entering directory '/home/wang/Desktop/openwrt-x86/package/libs/zlib'
echo "zlib" >> /home/wang/Desktop/openwrt-x86/staging_dir/target-i386_pentium4_glibc/pkginfo/zlib.default.install
make[2]: Leaving directory '/home/wang/Desktop/openwrt-x86/package/libs/zlib'
time: package/libs/zlib/compile#0.14#0.03#1.26
make[2]: Entering directory '/home/wang/Desktop/openwrt-x86/package/libs/gettext'
make[2]: Leaving directory '/home/wang/Desktop/openwrt-x86/package/libs/gettext'
time: package/libs/gettext/compile#0.13#0.04#0.41
make[2]: Entering directory '/home/wang/Desktop/openwrt-x86/package/libs/libiconv'
make[2]: Leaving directory '/home/wang/Desktop/openwrt-x86/package/libs/libiconv'
time: package/libs/libiconv/compile#0.15#0.03#0.34
make[2]: Entering directory '/home/wang/Desktop/openwrt-x86/package/devel/binutils'
make[2]: Leaving directory '/home/wang/Desktop/openwrt-x86/package/devel/binutils'
time: package/devel/binutils/compile#0.19#0.34#7.36
make[2]: Entering directory '/home/wang/Desktop/openwrt-x86/package/libs/libtool'
make[2]: Leaving directory '/home/wang/Desktop/openwrt-x86/package/libs/libtool'
time: package/libs/libtool/compile#0.12#0.05#0.54
make[2]: Entering directory '/home/wang/Desktop/openwrt-x86/feeds/packages/devel/gcc'
rm -f /home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/.built
touch /home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/.built_check
export SHELL="bash"; CFLAGS="-Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0:gcc-5.4.0 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/wang/Desktop/openwrt-x86/staging_dir/target-i386_pentium4_glibc/usr/include -I/home/wang/Desktop/openwrt-x86/staging_dir/target-i386_pentium4_glibc/include -I/home/wang/Desktop/openwrt-x86/staging_dir/toolchain-i386_pentium4_gcc-7.3.0_glibc/usr/include -I/home/wang/Desktop/openwrt-x86/staging_dir/toolchain-i386_pentium4_gcc-7.3.0_glibc/include -D_GLIBCXX_INCLUDE_NEXT_C_HEADERS " CXXFLAGS="-Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0:gcc-5.4.0 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/wang/Desktop/openwrt-x86/staging_dir/target-i386_pentium4_glibc/usr/include -I/home/wang/Desktop/openwrt-x86/staging_dir/target-i386_pentium4_glibc/include -I/home/wang/Desktop/openwrt-x86/staging_dir/toolchain-i386_pentium4_gcc-7.3.0_glibc/usr/include -I/home/wang/Desktop/openwrt-x86/staging_dir/toolchain-i386_pentium4_gcc-7.3.0_glibc/include -D_GLIBCXX_INCLUDE_NEXT_C_HEADERS " LDFLAGS="-L/home/wang/Desktop/openwrt-x86/staging_dir/target-i386_pentium4_glibc/usr/lib -L/home/wang/Desktop/openwrt-x86/staging_dir/target-i386_pentium4_glibc/lib -L/home/wang/Desktop/openwrt-x86/staging_dir/toolchain-i386_pentium4_gcc-7.3.0_glibc/usr/lib -L/home/wang/Desktop/openwrt-x86/staging_dir/toolchain-i386_pentium4_gcc-7.3.0_glibc/lib -znow -zrelro " make -C /home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0 DESTDIR="/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/ipkg-install"  all install
make[3]: Entering directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0'
make[4]: Entering directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0'
make[5]: Entering directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/libiberty'
make[6]: Entering directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/libiberty/testsuite'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/libiberty/testsuite'
make[5]: Leaving directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/libiberty'
make[5]: Entering directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/fixincludes'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/fixincludes'
make[5]: Entering directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/gmp'
make  all-recursive
make[6]: Entering directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/gmp'
Making all in tests
make[7]: Entering directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/gmp/tests'
Making all in .
make[8]: Entering directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/gmp/tests'
make[8]: Nothing to be done for 'all-am'.
make[8]: Leaving directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/gmp/tests'
Making all in devel
make[8]: Entering directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/gmp/tests/devel'
make[8]: Nothing to be done for 'all'.
make[8]: Leaving directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/gmp/tests/devel'
Making all in mpn
make[8]: Entering directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/gmp/tests/mpn'
make[8]: Nothing to be done for 'all'.
make[8]: Leaving directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/gmp/tests/mpn'
Making all in mpz
make[8]: Entering directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/gmp/tests/mpz'
make[8]: Nothing to be done for 'all'.
make[8]: Leaving directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/gmp/tests/mpz'
Making all in mpq
make[8]: Entering directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/host-i486-openwrt-linux-gnu/gmp/tests/mpq'
[... 160 lines cut ...]
In file included from ../.././libgcc/unwind-dw2.c:401:0:
./md-unwind-support.h: In function 'x86_fallback_frame_state':
./md-unwind-support.h:141:18: error: field 'uc' has incomplete type
  struct ucontext uc;
                  ^~
../.././libgcc/shared-object.mk:14: recipe for target 'unwind-dw2.o' failed
make[5]: *** [unwind-dw2.o] Error 1
make[5]: Leaving directory '/home/wang/Desktop/openwrt-x86/build_dir/target-i386_pentium4_glibc/gcc-5.4.0/i486-openwrt-linux-gnu/libgcc'
Makefile:12262: recipe for target 'all-target-libgcc' failed
make[4]: *** [all-target-libgcc] Error 2
[... cut ...]
make[2]: Leaving directory '/home/wang/Desktop/openwrt-x86/feeds/packages/devel/gcc'
Command exited with non-zero status 2
time: package/feeds/packages/gcc/compile#1.49#1.08#19.61
package/Makefile:107: recipe for target 'package/feeds/packages/gcc/compile' failed
make[1]: *** [package/feeds/packages/gcc/compile] Error 2
make[1]: Leaving directory '/home/wang/Desktop/openwrt-x86'
/home/wang/Desktop/openwrt-x86/include/toplevel.mk:216: recipe for target 'package/gcc/compile' failed
make: *** [package/gcc/compile] Error 2
wang@wang-VirtualBox:~/Desktop/openwrt-x86$

dibdot, 2018-10-16:
@noblepepper

805447391, 2018-10-25:
OK, i have solved this problem temporary. Here is a very ugly solution.
File:{openwrt-ROOT}/build_dir/target-x86_64-glibc/gcc-5.4.0/x86_64-openwrt-linux-gnu/libgcc/md-unwind-support.h
Fix1:@line 61:
if (*(unsigned char *)(pc+0) == 0x48
      && *(unsigned long long *)(pc+1) == RT_SIGRETURN_SYSCALL)
    {
      // struct ucontext *uc_ = context->cfa;
	ucontext_t *uc_ = context->cfa;
      /* The void * cast is necessary to avoid an aliasing warning.
         The aliasing warning is correct, but should not be a problem
         because it does not alias anything.  */
      sc = (struct sigcontext *) (void *) &uc_->uc_mcontext;
    }
Fix2:@line144
    {
      struct rt_sigframe {
	int sig;
	siginfo_t *pinfo;
	void *puc;
	siginfo_t info;
	// struct ucontext uc;
 	ucontext_t uc;
      } *rt_ = context->cfa;

See https://github.com/voidlinux/void-packages/issues/7284

pprindeville, 2018-10-28:
I suggest doing a PR instead of a pseudo-fix in the comments of an issue.

noblepepper, 2019-02-19:
The package desciption  says: this was only tested on arm (EABI) and mips targets. Others to be
done...

On a mips build md-unwind-support.h is different from what you quote so I can't test your fix. If you get a working gcc and can supply a patch to the sources I'll try to integrate into this package.

neheb, 2020-01-12:
This is not valid anymore. There's GCC7 now.

closed 2020-01-12 by neheb