openwrt/packages #7203: cloudflare.com-v4 subdomain update question

SUNJINZE, 2018-10-16:
Packages:ddns-scripts_cloudflare.com-v4
Luci has only three variables [PASSWORD], [DOMAIN] and [IP], and I want DDNS to update x.xx.com under the xxx.com domain name.
I hope you can add sub domain variables in Luci, and HOST=DOMAIN in the script now.
Thanks!

dibdot, 2018-10-16:
Seems to be a configuration issue - please ask in the forum, cause this package is unmaintained.

closed 2018-10-16 by dibdot

chris5560, 2018-10-24:
using following options from /etc/config/ddns
option username  - your cloudflare e-mail
option password  - cloudflare api key, you can get it from cloudflare.com/my-account/
option domain    - "hostname[.subdomain]@yourdomain.TLD"

this should do the job

SUNJINZE, 2018-10-24:
> 使用/ etc / config / ddns选项用户名中的以下选项 - 您的cloudflare电子邮件选项密码 - cloudflare api密钥，您可以从cloudflare.com/my-account/选项域获取 - “hostname [.subdomain] @yourdomain。 TLD“这应该做的工作

I found the reason because I filled in the hostname.yourdomain without adding @ so the update was unsuccessful
Thank you @chris5560