openwrt/packages #7210: libexif

mendel129, 2018-10-17:
Maintainer: @MikePetullo
Environment: OpenWrt 18.06

Description: could https://github.com/libexif/libexif/commit/c39acd1692023b26290778a02a9232c873f9d71a#diff-830e348923810f00726700b083ec00cd be added to the OpenWRT version of libexif?
They migrated to that GitHub in the meantime including some other fixes as well.
https://github.com/libexif/libexif 

It's the same issue as originally followed via https://sourceforge.net/p/libexif/bugs/130/ 

Thanks in advance!

diizzyy, 2018-10-19:
Perhaps you can do a pull request that uses the master branch instead?
You can use this as a template for using codeload on master branch
https://github.com/openwrt/packages/pull/6010/files#diff-93a681dc930689e08c33cee51a9aad1bR17

diizzyy, 2019-07-14:
Closing due to timeout

closed 2019-07-14 by diizzyy