openwrt/packages #7227: CONTRIBUTING.md -- squashing PR's into a single commit?

cshoredaniel, 2018-10-21:
Hi,

I always thought that commits to packages (or OpenWrt) were preferred to be minimal atomic commits (i.e. do one thing, should work standalone, except bumping PKG_RELEASE), but on at least one PR I was asked to squash a set of commits on a package into a single commit.

What is the packages feed policy on this, and should it be added to CONTRIBUTING.md file?

Is the difference that is was a backport to 18.06 of a selected set of commits?

pprindeville, 2018-10-21:
My understanding is the same as yours, assuming "do one thing" equates to "fix one issue" or "add one enhancement" at that granularity.

mhei, 2018-10-21:
I'd agree to this. However, there is always some room for interpretation, so I don't know who/why you was requested to squash the commits. I'd also prefer to see each individual backported commit as is, squashing at this point could make it harder to follow what was backported and what was modified during the backporting process.

EricLuehrsen, 2018-10-25:
As with everything the answer is "it depends." Considerations include the mess in front of you and your future plans to reuse it. If you have a major refactoring that has no chance of back porting cleanly anyway, then just squash it to one commit. If you have clean fixes that could be cherry-picked, then keep them separate.

jonathanunderwood, 2018-11-04:
I think better criteria would be:

- Only fix a single issue, or implement a single feature in a PR
- Each commit should be self contained and testable when compiled (so that git blame remains useful

I think insisting on squashing commits is unhelpful, particularly when multiple people are collaborating on a feature/fix: the requirement to squash implicitly assumes a single author.

closed 2018-12-06 by cshoredaniel