openwrt/packages #7228: seafile-seahub: fails to build due to missing build dependency

kissg1988, 2018-10-21:
Maintainer: @kissg1988 (adding @cotequeiroz as the most recent contributor)
Environment: ar71xx, openwrt HEAD (issue exists for all archs)

Description:

According to the [build log](https://downloads.openwrt.org/snapshots/faillogs/mipsel_74kc/packages/seafile-seahub/compile.txt) the build fails due to being unable to locate the pytz python module which is available as a [package](https://github.com/openwrt/packages/blob/master/lang/python/pytz/Makefile) and is a runtime dependency for the seafile-seahub package. Please someone take a look at it, I believe specifying pytz as a build dependency in PKG_BUILD_DEPENDS would fix the issue. Unfortunately, I'm not currently able to create the fix myself, hence reporting this as an issue.

Thanks!

```
ImportError: No module named pytz
Makefile:14: recipe for target 'locale' failed
make[4]: *** [locale] Error 1
make[4]: Leaving directory '/var/lib/buildbot/slaves/slave-lede-builds4/mipsel_74kc/build/sdk/build_dir/target-mipsel_74kc_musl/seafile-seahub-6.3.0'
Makefile:63: recipe for target '/var/lib/buildbot/slaves/slave-lede-builds4/mipsel_74kc/build/sdk/build_dir/target-mipsel_74kc_musl/seafile-seahub-6.3.0/.built' failed
make[3]: *** [/var/lib/buildbot/slaves/slave-lede-builds4/mipsel_74kc/build/sdk/build_dir/target-mipsel_74kc_musl/seafile-seahub-6.3.0/.built] Error 2
```

cotequeiroz, 2018-10-24:
Sorry for not getting back to you earlier.  I will take a look at it, and see what I can do.

cotequeiroz, 2018-10-26:
In my system pytz always gets built before seafile-seahub.  I've tried it with `CONFIG_BUILDBOT=y`  to emulate the bots, but I'm unable to reproduce the problem.  I'll open a PR to see if travis has a similar problem, and see if the PKG_BUILD_DEPENDS fixes it.

cotequeiroz, 2018-10-29:
I've tried you suggestion, but it did not work.  I found a way to get an error locally now, but it is in seafile-ccnet.  If I get the config.seed from the buildbot and select seanet-server, `make -j3` fails. If I run it again, it builds.  It takes forever to try again, though.  I'll study it some more, and keep you posted.

kissg1988, 2018-10-29:
Not sure what could be wrong here, unreproducible build issues are always nasty. :(

As a quick tip, it's worth checking whether pytz installs to the correct location and/or PYTHONPATH is correctly set during the build process (most likely it is). I don't see a faillog for pytz so it should be built and installed to its expected location, yet it's still not available. Maybe trying a clean rebuild of the package on the buildbot system would help.

cotequeiroz, 2018-10-31:
You were right: pytz is missing InstallDev. I guess seafile finds pytz at pytz's build dir, but the bots clean it after building it, and seafile-seahub fails.  I'll open a PR right away to see if travis builds it or not.  I suspect travis has another issue as pytz is not even on the list when it fails.

closed 2018-11-01 by champtar