openwrt/packages #7230: [mariadb] SIGSEGV during access

HeadLessHUN, 2018-10-21:
Maintainer: Sebastian Kemper
Environment: mips32r2, mir3g, r8353-d27d87c
Package version: mariadb-server - 10.2.17-1 (built on 2018.10.20.)

Description:

 -Start mysql server (mariadb)
 -Attempt to connect to the server through socket or tcp.
 -server stop because of invalid memory write 

```
Sun Oct 21 13:22:41 2018 kern.info kernel: [ 1807.502009] do_page_fault(): sending SIGSEGV to mysqld for invalid write access to 00000000
Sun Oct 21 13:22:41 2018 kern.info kernel: [ 1807.510442] epc = 77f02490 in libc.so[77e88000+95000]
Sun Oct 21 13:22:41 2018 kern.info kernel: [ 1807.515582] ra  = 77f023d8 in libc.so[77e88000+95000]
```

diizzyy, 2018-10-21:
Can you try getting global CFLAGS to -O2 instead of -Os and do a full recompile?
There are reports where -Os miscompiles on mips so it might be worth testing as a quick fix.

HeadLessHUN, 2018-10-21:
I did not mention that before but the default Target/Package CFLAGS are already O2. In the makefile i don't see any overwrites, so it should be O2.

Now I'm trying to compile with glibc instead of musl, but that is not permanent solution.

Maybe it is connected with [this?](https://forum.openwrt.org/t/mariadb-illegal-instruction/21984)

I do not remember clearly but i think i built my previous image on 09.06. and it was working perfectly. So as i see git nothing has changed to mariadb since then. 

You have any other thoughts?

HeadLessHUN, 2018-10-22:
I temporarily solved my issue.
I tried to build with glibc, that wouldn't be a good solution and didn't built. So i checkout the 18.06 branch.. Here is the problem, my database was used with 10.2.17 as i remembered i built previous image on 2018.09.06. So mysql could'nt use that database i didn't want to roll back to previous backup it was one week old. So i modified the feeds.conf to use master update/install all feeds. built image now with 10.2.17, it successfully started and i can connect. so it must be something else, not mysql related. Maybe the musl update or device specific kernel config modification which was in this time intervall.

closed 2018-10-22 by HeadLessHUN

micmac1, 2018-11-10:
Hello @HeadLessHUN 

I asked on the musl mailing list for help. Here's the reply: [link](https://www.openwall.com/lists/musl/2018/11/10/8)

Kind regards,
Seb