openwrt/packages #7280: dnsmasq: Assigning the same static IP address to two different hosts breaks everything

witten, 2018-10-30:
# Issue template (remove lines from top till here)

Maintainer: Kevin Darbyshire-Bryant (could not find GitHub user)
Environment: TP-Link Archer C7 v2, openwrt-18.06 branch (git-18.228.31946-f64b152) / OpenWrt 18.06.1 r7258-5eb055306f 

Description:

When configuring static leases in LuCI's UI, I accidentally assigned the same IPv4 IP address to two different devices with different hostnames and different MAC addresses. When I hit save, dnsmasq completely stopped responding. This manifested as no DNS responses, and no DHCP addresses allocated.

The only way I was able to fix the problem was to manually assign a client machine an IP address with ifconfig, reconnect to LuCI, and correct the errant IP address. When I did that and hit save, dnsmasq started doing both DNS and DHCP again.

Unfortunately I don't have any logs from when this happened. But if you tell me where to look, I can try to repro again and dig up the relevant logs.

Expected behavior:

When a duplicate IP address is entered, the ideal would be for LuCI to prevent that with a nice error message. (I can file a separate ticket for that if you think it's a good idea.) But short of that, dnsmasq should probably not stop serving DNS and DHCP when given an incorrect configuration like the one described.

ldir-EDB0, 2018-10-30:
I'm at the openwrt conference at the moment but when I get back I'll take a look.

ldir-EDB0, 2018-10-30:
ok, so I had a very quick look at this.  In the scenario you describe, luci will restart dnsmasq and dnsmasq will throw an error on the resulting config file because duplicate host IP addresses are a bad idea:

Tue Oct 30 15:12:15 2018 daemon.crit dnsmasq[9477]: duplicate dhcp-host IP address 192.168.219.14 at line 47 of /var/etc/dnsmasq.conf.cfg01411c
Tue Oct 30 15:12:15 2018 daemon.crit dnsmasq[9477]: FAILED to start up

Ideally luci wouldn't produce this as an illegal configuration and that needs to be fixed first. Maybe dnsmasq should be more tolerant of this scenario and 'shout loudly' but not abort.  Patches welcome

I'll have a chat.  Patches welcome :-)

jow-, 2018-10-30:
Duplicate of https://github.com/openwrt/luci/issues/1079 - will add a fix to LuCI later.

closed 2018-10-30 by jow-