openwrt/packages #7283: collectd-mod-openvpn: does not create all luci-statistics needed files.

velainur, 2018-10-30:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (x86, generic, OpenWrt 18.06.1)

Firstly, i know that there are some other post similar to this one, but without any solution, so i think it was better to start a clean one instead. (since they are so old, and maybe file version dont match with mines).

This is the (UCI) config related the custom "status" file for openvpn:

```
openvpn.MyOVPN.status='/mnt/OpenVPN/MyOVPN 10'
luci_statistics.collectd_openvpn.StatusFile='/mnt/OpenVPN/MyOVPN'
```
firstly, i noticed that the file which has been created is created without world-readable permissions. So i had to enable them via chmod command.

Then, i found that several rrd files was created:

```
root@MainRouter:/mnt/Collectd/MainRouter/openvpn-MyOVPN# ls -lah
drwxr-xr-x    2 root     root        4.0K Oct 19 11:51 .
drwxr-xr-x   32 root     root        4.0K Oct 19 11:49 ..
-rw-r--r--    1 root     root       17.5K Oct 19 11:51 if_octets-Client1.rrd
-rw-r--r--    1 root     root        9.3K Oct 19 11:51 users-MyOVPN.rrd
```
This is the configuration of the collectd-mod-openvpn:
```
luci_statistics.collectd_openvpn=statistics
luci_statistics.collectd_openvpn.enable='1'
luci_statistics.collectd_openvpn.CollectIndividualUsers='1'
luci_statistics.collectd_openvpn.CollectCompression='1'
luci_statistics.collectd_openvpn.ImprovedNamingSchema='1'
luci_statistics.collectd_openvpn.CollectUserCount='1'
luci_statistics.collectd_openvpn.StatusFile='/mnt/OpenVPN/MyOVPN'
```
I think that this isnt good enough since when i try to check the graph inside luci-app-statistics, i found this log entries:
```
Fri Oct 19 11:52:41 2018 daemon.err uhttpd[2969]: ERROR: opening '/mnt/
Collectd/MainRouter/openvpn-MyOVPN/if_octets-traffic.rrd': No such file or directory 
Fri Oct 19 11:52:41 2018 daemon.err uhttpd[2969]: ERROR: opening '/mnt/
Collectd/MainRouter/openvpn-MyOVPN/compression-data_out.rrd': No such file or directory
```
So I guess collectd-mod-openvpn isnt collecting all stats, needed.
Any clue?

Regards,

velainur, 2018-11-05:
no clues?

velainur, 2018-11-15:
Help, pls!

jlagden, 2019-03-03:
Something seems buggy/broken.

I'm running 18.06.2 and get these messages:

```
Sun Mar  3 21:59:23 2019 daemon.err uhttpd[1841]: ERROR: opening '/tmp/rrd/openwrt/openvpn-openvpn-status.log/if_octets-traffic.rrd': No such file or directory
Sun Mar  3 21:59:23 2019 daemon.err uhttpd[1841]: ERROR: opening '/tmp/rrd/openwrt/openvpn-openvpn-status.log/compression-data_out.rrd': No such file or directory
Sun Mar  3 22:00:13 2019 daemon.err uhttpd[1841]: ERROR: opening '/tmp/rrd/openwrt/openvpn-openvpn-status.log/if_octets-traffic.rrd': No such file or directory
Sun Mar  3 22:00:13 2019 daemon.err uhttpd[1841]: ERROR: opening '/tmp/rrd/openwrt/openvpn-openvpn-status.log/compression-data_out.rrd': No such file or directory
```

There were no options in luci after initially installing until I added a basic config using uci.

Then, the plugin wouldn't record any rrd files until I selected the collectindividualusers and improvednamingschema options.

```
root@openwrt:~# uci show luci_statistics.collectd_openvpn
luci_statistics.collectd_openvpn=statistics
luci_statistics.collectd_openvpn.enable='1'
luci_statistics.collectd_openvpn.StatusFile='/tmp/openvpn-status.log'
luci_statistics.collectd_openvpn.CollectIndividualUsers='1'
luci_statistics.collectd_openvpn.ImprovedNamingSchema='1'
```

jlagden, 2019-03-04:
Had a quick look at lunchtime. Depending on the OpenVPN mode the status file format is different:
```
 * There is two main kinds of OpenVPN status file:
 * - for 'single' mode (point-to-point or client mode)
 * - for 'multi' mode  (server with multiple clients)
```
The traffic and compression rrd files only get produced if it is a single mode file.
In multi-mode you end up with a file or files with the client common name eg. `if_octets-Client1.rrd1`

luci statistics is not setup to display the rrd files produced in multi mode:

[https://github.com/openwrt/luci/blob/openwrt-18.06/applications/luci-app-statistics/luasrc/statistics/rrdtool/definitions/openvpn.lua](https://github.com/openwrt/luci/blob/openwrt-18.06/applications/luci-app-statistics/luasrc/statistics/rrdtool/definitions/openvpn.lua)

velainur, 2019-03-07:
So... How can we make it works?

I have some programming knowledge, but i dont know how it really works. Is there any documentation so i can learn how this kind of plugins works? i would love to fix it if i can.

Regards.

feckert, 2019-05-25:
Have a look into https://collectd.org/wiki/index.php/Plugin:OpenVPN and
And have also a look at the description for luci_statistics https://github.com/openwrt/luci/tree/master/applications/luci-app-statistics/luasrc/statistics/rrdtool/definitions there are some examples that you can use as a template.

Menion2k, 2021-01-14:
Hello

The problema apparently is still pending. The issue is with mismatch of what collectd creates and what luci-statistics wants
In fact if I copy the rdd from a client openvpn (working) to multi server (not working) it creates the graph.

It is not only a matter of file name, but also for clients, rdd creates one file **if_octets-Client1.rrd**, while clients create two files: **if_octets-traffic.rrd** and **if_octets-overhead.rrd**

I am not sure if the problem shall be fixed on luci-statistics (I think) or on collectd side?