openwrt/packages #7285: v4l-utils:

xabolcs, 2018-10-30:
### Maintainer: @thess 

### Environment:
```
root@OpenWrt:~# cat /etc/os-release 
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="http://openwrt.org/"
BUG_URL="http://bugs.openwrt.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r8391-162e016"
LEDE_BOARD="ipq40xx/generic"
LEDE_ARCH="arm_cortex-a7_neon-vfpv4"
LEDE_TAINTS=""
LEDE_DEVICE_MANUFACTURER="OpenWrt"
LEDE_DEVICE_MANUFACTURER_URL="http://openwrt.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="OpenWrt SNAPSHOT r8391-162e016"
```

```
root@OpenWrt:~# cat /tmp/sysinfo/model 
GL.iNet GL-B1300
```
---

### Description:

I have a little USB FM Transmitter with `046d:0a0e` IDs.

It has a nice controller program, called [`keene`](https://github.com/kenchy/keene-usb-audio) which I was able to refresh it's OpenWrt support.

But nowadays there is `v4l-ctl`, which is able to control the transmitter on my desktop environment:  
```
$ v4l2-ctl --list-devices
Keene FM Transmitter (usb-0000:00:14.0-13.3):
	/dev/radio0
```

Example usage:
```
$ v4l2-ctl --device=/dev/radio0 --set-modulator=stereo
$ v4l2-ctl --device=/dev/radio0 --get-modulator
Modulator 0:
	Name                 : FM
	Type                 : radio
	Capabilities         : 62.5 Hz stereo freq-bands 
	Frequency range      : 76.0 MHz - 108.0 MHz
	Subchannel modulation: stereo 
```

Sadly on OpenWrt `v4l2-ctl` misses `/dev/video` very hard, even for `--list-devices`:

```
root@OpenWrt:~# v4l2-ctl --list-devices
Failed to open /dev/video0: No such file or directory
```

lsusb:
```
root@OpenWrt:~# lsusb 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 046d:0a0e Logitech, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I would like to control my Keene FM Transmitter from my OpenWrt box! :blush: 

It's OK to me to introduce that `keene` package as a last resort.

Sizes:
```
root@OpenWrt:~# ls -l keene_20181029-1_arm_cortex-a7_neon-vfpv4.ipk 
-rw-r--r--    1 root     root          3861 Oct 30 02:57 keene_20181029-1_arm_cortex-a7_neon-vfpv4.ipk

root@OpenWrt:~# ls -l /usr/bin/keene 
-rwxr-xr-x    1 root     root          8337 Oct 29 15:20 /usr/bin/keene
```

Thank You for helping!

rafaeldomi, 2018-11-06:
$00$0 p$0xp,p0$0xp$0,p,p

thess, 2018-11-10:
I think you first need to get your device recognized by the system with an appropriate driver.
The v4l2-ctl error message "Failed to open /dev/video0: No such file or directory" is just its way of saying it found NO compatible devices. Also, you probably don't have a /dev/radio0 device either as you see on your PC.

First step is to install the driver(s) you need. From what I find on the web about your device here:
http://blog.palosaari.fi/2013/08/naked-hardware-12-usb-audio-transmitter.html?showComment=1540228540725

You will at least need to install USB Audio support (**kmod-usb-audio**) to get the audio portion of the device seen. The device also has an HID interface so you will need at a minimum **kmod-usb-hid** and possibly **kmod-hid-common**. 

All that said, since I don't have one of these devices, you'll have to try things out yourself.

thess, 2018-11-10:
Going forward - perhaps you should also ask for help to configure your device on our forum (https://forum.openwrt.org) since AFAICT, this is not a bug in v4l-utils. There are more people there to help with such issues.

closed 2018-11-10 by thess

xabolcs, 2018-11-11:
Thanks for the link!

I suspected that there is a missing driver, but I had no clue which one. But your link helped:
> ... Kernel driver name is radio-keene, done by Norwegian V4L superman Hans Verkuil. ...

Desktop log:
```
Nov 11 14:35:33 mch kernel: usb 2-4.2: new full-speed USB device number 8 using xhci_hcd
Nov 11 14:35:33 mch kernel: usb 2-4.2: New USB device found, idVendor=046d, idProduct=0a0e, bcdDevice= 1.00
Nov 11 14:35:33 mch kernel: usb 2-4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Nov 11 14:35:33 mch kernel: usb 2-4.2: Product: B-LINK USB Audio  
Nov 11 14:35:33 mch kernel: usb 2-4.2: Manufacturer: HOLTEK 
Nov 11 14:35:33 mch mtp-probe[5142]: checking bus 2, device 8: "/sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4.2"
Nov 11 14:35:33 mch kernel: radio-keene 2-4.2:1.2: V4L2 device registered as radio0
Nov 11 14:35:33 mch mtp-probe[5142]: bus: 2, device: 8 was not an MTP device
```

I checked the `kmod`s, but did not found any related group, only `kmod-video-*`.

So my problem would be fixed with introducing the `kmod-radio-keene` package, which would be the first `kmod-radio-*` package?