openwrt/packages #7289: minidlna failing to build

davidc502, 2018-10-31:
minidlna is failing to build. I've noticed it is also failing to build in the openwrt daily snapshots.

This is the error I'm seeing:

[code]
configure: error: libavutil headers not found or not usable
Makefile:83: recipe for target '/home/davidc502/Desktop/lede-build/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/minidlna-1.2.1/.configured_889556d2f423f99e091beece9c8d870a' failed
make[3]: *** [/home/davidc502/Desktop/lede-build/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/minidlna-1.2.1/.configured_889556d2f423f99e091beece9c8d870a] Error 1
make[3]: Leaving directory '/home/davidc502/Desktop/lede-build/openwrt/feeds/packages/multimedia/minidlna'
time: package/feeds/packages/minidlna/compile#6.38#1.78#9.48
    ERROR: package/feeds/packages/minidlna failed to build.
[/code]

diizzyy, 2018-11-02:
This is most likely because it uses libffmpeg as dependency which isn't a package however it is defined.
https://github.com/openwrt/packages/blob/master/multimedia/ffmpeg/Makefile#L738

I tried compiling minidlna against ffmpeg40 which works fine
https://github.com/diizzyy/packages/commit/b94e499ac6ef6aa5f7cfbd8ca852cae4cde8105b

You can use it if you want (works fine on mvebu) however you need to drop libx264, fdk-aac and "--non-free" if you want to distribute it. You can also use the -custom variant in the tree however I haven't check what you need to enable to make it usable.

thess, 2018-11-02:
This is a known problem with the way CONFIG dependencies work and soft vs hard float. For building minidlna, selecting libffmpeg-mini should work for you. That was the intended use for -mini.
I am working on a solution for libffmpeg-full on hard float but haven't been successful yet.

davidc502, 2018-11-02:
Thanks, I saw it was associated with ffmpeg and was going to do Full, but will go ahead and try -mini instead.

Appreciate the research.

thess, 2018-11-12:
c44e1d43a36343564972b48809357fa7cef3fa5c resolves ambiguities with BUILD_PATENTED and hard/soft float options in libffmpeg. Closing this,

closed 2018-11-12 by thess