openwrt/packages #7303: rng-tools: outdated, version 6.6 is available

gnustomp, 2018-11-03:
Maintainer: @nwf 
Environment: 18.06.1, x86_64 (but widely applicable)

Description:

rng-tools now lives at https://github.com/nhorman/rng-tools , and 6.6 is available.

There are also some new entropy sources for rngd to use, notably `jitterentropy`.
It may be desirable to use this instead of `/dev/urandom` for additional entropy.

nwf, 2018-11-03:
Thanks for the heads up.  I'm away from my openwrt build system right now, but I'll package the new version when I get a chance.

How does `jitterentropy` compare to havege?  Do I read right that `jitterentropy` requires a kernel module?  Should we deprecate the havege packages if `jitterentropy` supersedes it?  Is there harm in having both installed at the same time? :)

hnyman, 2018-11-04:
> havege being stale. Looks like a new upstream maintainer, not just a change of host.

Hmmm. Apparently legit new site for haveged, as the old site mentions new github development in the page http://www.issihosts.com/haveged/

nwf, 2018-11-10:
This can be closed now, I think.

closed 2018-11-11 by hnyman