openwrt/packages #7305: ldns: build fails

simontretter, 2018-11-04:
Maintainer: @neheb @damianorenfer 
Environment: ramips_mt7621, WG3526, OpenWrt trunk)

Description:
ldns build is fails (gcc 7).

```
./dnssec.c:1801:2: error: unknown type name 'DSA_SIG'; did you mean 'ECDSA_SIG'?
  DSA_SIG *dsasig;
  ^~~~~~~
  ECDSA_SIG
./dnssec.c:1808:4: warning: assignment makes pointer from integer without a cast [-Wint-conversion]                                                                                  
  R = BN_new();
    ^
./dnssec.c:1812:4: warning: assignment makes pointer from integer without a cast [-Wint-conversion]                                                                                  
  S = BN_new();
    ^
./dnssec.c:1820:11: warning: implicit declaration of function 'DSA_SIG_new'; did you mean 'ECDSA_SIG_new'? [-Wimplicit-function-declaration]                                         
  dsasig = DSA_SIG_new();
           ^~~~~~~~~~~
           ECDSA_SIG_new
./dnssec.c:1820:9: warning: assignment makes pointer from integer without a cast [-Wint-conversion]                                                                                  
  dsasig = DSA_SIG_new();
         ^
./dnssec.c:1830:8: error: request for member 'r' in something not a structure or union
  dsasig->r = R;
        ^~
./dnssec.c:1831:8: error: request for member 's' in something not a structure or union
  dsasig->s = S;
        ^~
./dnssec.c:1834:16: warning: implicit declaration of function 'i2d_DSA_SIG'; did you mean 'i2d_ECDSA_SIG'? [-Wimplicit-function-declaration]                                         
  raw_sig_len = i2d_DSA_SIG(dsasig, &raw_sig);
                ^~~~~~~~~~~
                i2d_ECDSA_SIG
make[4]: *** [Makefile:122: dnssec.lo] Error 1
make[4]: Leaving directory '/home/deploy/openwrt/WG3526/build_dir/target-mipsel_24kc_musl/ldns-1.7.0'                                                                                
make[3]: *** [Makefile:87: /home/deploy/openwrt/WG3526/build_dir/target-mipsel_24kc_musl/ldns-1.7.0/.built] Error 2                                                                  
make[3]: Leaving directory '/home/deploy/openwrt/WG3526/feeds/packages/libs/ldns'
time: package/feeds/packages/ldns/compile#49.11#11.85#65.03
make[2]: *** [package/Makefile:109: package/feeds/packages/ldns/compile] Error 2
make[2]: Leaving directory '/home/deploy/openwrt/WG3526'
make[1]: *** [package/Makefile:103: /home/deploy/openwrt/WG3526/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2                                                  
make[1]: Leaving directory '/home/deploy/openwrt/WG3526'
make: *** [/home/deploy/openwrt/WG3526/include/toplevel.mk:218: world] Error 2
```

neheb, 2018-11-04:
https://github.com/openwrt/packages/pull/7300

Enable deprecated APIs.

simontretter, 2018-11-04:
That indeed fixes things.

closed 2018-12-02 by simontretter