openwrt/packages #7309: snmpd: ipAdEntIfIndex: shows wrong interface index

fennsen64, 2018-11-04:
# Issue template (remove lines from top till here)

Maintainer: @\Stijn Tintel (find it by checking history of the package Makefile)
Environment: (put here MIPSBE, AR7161, OpenWrt 18.06)

Description:
The value of some ipAdEntIfIndex are pointing to wrong interface index values.
With this my opennms management engine gets silly :-(

Example here:
Interface with IP 172.16.0.1 which sits on eth1.1 (ifName.30) is pointed to loopback interface (ifName.1)

```# snmpwalk -c public -v2c localhost ipAdEntIfIndex
IP-MIB::ipAdEntIfIndex.127.0.0.1 = INTEGER: 1
IP-MIB::ipAdEntIfIndex.**172.16.0.1 = INTEGER: 1**
IP-MIB::ipAdEntIfIndex.188.110.255.182 = INTEGER: 32
IP-MIB::ipAdEntIfIndex.192.168.0.10 = INTEGER: 2
root@OpenWrt:~# snmpwalk -c public -v2c localhost ifName
IF-MIB::ifName.1 = STRING: lo
IF-MIB::ifName.2 = STRING: eth0
IF-MIB::ifName.3 = STRING: eth1
IF-MIB::ifName.4 = STRING: ifb0
IF-MIB::ifName.5 = STRING: ifb1
IF-MIB::ifName.29 = STRING: br-lan
IF-MIB::ifName.30 = STRING: eth1.1
IF-MIB::ifName.31 = STRING: eth1.2
IF-MIB::ifName.32 = STRING: pppoe-wan2
IF-MIB::ifName.36 = STRING: ifb4eth0
```

Seems to be are very old bug; should already been fixed since net-snmp 5.1.2 upwards

Patch / History See here:
https://bugzilla.redhat.com/show_bug.cgi?id=119106
https://bugzilla.redhat.com/attachment.cgi?id=98840&action=diff

Same issue here:
https://sourceforge.net/p/net-snmp/mailman/message/35807478/
```
Formating code blocks by wrapping them with pairs of ```
```

karlp, 2018-11-04:
So, just to be sure, what version of net-snmp are you actualyl running?

fennsen64, 2018-11-05:
NET-SNMP version:  5.7.3

karlp, 2018-11-06:
best I can suggest is to talk to upstream about it.  we're not using any legacy old kernels here.

fennsen64, 2018-11-07:
what do you mean with old kernels? I'm using:
OpenWrt 18.06 (the actual kernel) and latest SNMP version (5.7.3) from the distro

The provided links are only a hint that this problem seems to be quite old.

fennsen64, 2018-11-08:
ah ok, thanks for the clarification.
Yes it seems to be a new bug; what do you mean with 'speak to upstream'? 
Sorry I'm not very familar with the development processes / taxonmy used here as I'm not a developer.
I guess you mean that the net-snmp bug is not an openwrt release issue, but a net-snmp bug which impacts all distros, correct?
So: what do I have to do to adress the right channel that this bug will get fixed?

karlp, 2018-11-08:
file a bug in that same bugzilla you linked earlier, or try posting to that mailing list.  Include any and all information you can on how to reproduce it.  I have a simplet setup here, but the indcies all match on my device.

stintel, 2019-02-16:
@fennsen64 did you report this upstream? If so please link the report here.