openwrt/packages #7324: <python3-cryptography> missing libs *.so

kofec, 2018-11-06:
Maintainer: @jefferyto @commodo 
Environment: Raspberry Pi, arm_arm1176jzf-s_vfp,

Description:
It looks looks like ther is no files *.so in directory which seems to be required:
```
ls -lah /usr/lib/python3.6/site-packages/cryptography/hazmat/bindings/
drwxr-xr-x    3 root     root        4.0K Nov  6 11:19 .
drwxr-xr-x    5 root     root        4.0K Nov  6 11:15 ..
-rw-r--r--    1 root     root         246 Oct 29 10:56 __init__.py
-rw-r--r--    1 root     root         189 Nov  6 11:19 __init__.pyc
-rwxr-xr-x    1 root     root        4.0K Nov  6 11:19 _constant_time.abi3.so
-rwxr-xr-x    1 root     root      690.8K Nov  6 11:19 _openssl.abi3.so
-rwxr-xr-x    1 root     root        4.0K Nov  6 11:19 _padding.abi3.so
```

commodo, 2018-11-06:
Will check.
Assign it to me if you can

commodo, 2018-11-14:
@kofec 
Not sure what changed, I guess there was maybe an intermittent failure somewhere [in the build].
But now the libraries are present.

I checked the archive:  https://downloads.openwrt.org/snapshots/packages/arm_arm1176jzf-s_vfp/packages/python3-cryptography_2.3.1-1_arm_arm1176jzf-s_vfp.ipk

closed 2018-11-21 by kofec